thonny/thonny #1668: Thonny reporting unexpected echo.

nickwebcouk, 2021-03-01:
I'm currently using Thonny with a Pi Pico and I'm working on some GPS code which uses the RX and TX Pins. Part of my code sets ups a UART and prints what it receives to the shell space at the bottom.

Every now and then, when I make some code changes and attempt to run the program I get the error at the bottom forcing me to restart.

The text after the b` relates to what I had been echoing. The error is:

`WARNING:root:Unexpected echo. Expected b'%Run -c $EDITOR_CONTENT\r\n', got b'Sat in view:04 Valid: 1 C'`

My code is simply:

`from machine import UART,Pin
import utime

uart = UART(0,baudrate=9600, tx=Pin(0), rx=Pin(1), bits=8, parity=None, stop=1)

while True:
    try:
        if uart.any():
            rcvChar = uart.readline()
            #print(rcvChar.decode("ascii"),end="")
            line = rcvChar.decode("ascii")
            if "$GPGGA" in line:
                fixData = line.split(",")
                print("Sat in view:" + fixData[7] + " Valid: " + fixData[6] + " Current Time: " + fixData[1] + "(" + str(utime.localtime()) + ")")
                #neededData = (float(line[2]), line[3], float(line[4]), line[5], float(line[9]))
                #print(neededData)
            if "$GPVTG" in line:
                speed = line.split(",")
                kph = speed[5]
                mph = str(float(speed[5]) * 0.621371)
                print("Speed: " + kph + "km/h, " + mph + "MPH")
    except:
        print("Awaiting valid signal")
        #print(line)
`

Is this something I Can combat with my code or is this an issue with Thonny itself?

[image]

aivarannamaa, 2021-03-01:
Thank you for the report!

It seems to be Thonny's fault. The Shell seems to think that `%Run -c $EDITOR_CONTENT` is the input for your program on Pico, while it's actually a marker created by the Run command.

On your screenshot you have two horizontal lines after your program's output. Can you tell me what happened there? Did you invoke Run command several times in a row?

Until I'm finding and fixing the problem I can suggest interrupting the program with Ctr+C (so that you'll get `>>> ` prompt) before trying to run it again.

nickwebcouk, 2021-03-06:
Managed to recreate this today. Running the code above (well, almost - had a few changes to it) and working my way through various errors (not escaping strings correctly, missing : on statements etc) with the usual red errors.

Hit the run button and code runs fine. Make another code change and press the run button again to update the running code and the error above appeared.

Pressing CTRL+C resets it and allows me to press run again with no issues.

I have a feeling that when I'm hitting return because I'm outputting so much to the shell that comes being thrown somewhere it shouldn't.

Hope that helps!

aivarannamaa, 2021-03-08:
Thank you for the information!

I'll try to reproduce it later. From there on it should be easy :)

tavasjn, 2021-03-30:
Were either of you able to fix this issue? I have been running into the same issue I am also running Thonny on Mac M1 Big Sur. If you have a solution to this please share.

aivarannamaa, 2021-04-09:
I haven't had time to look into this yet.

Jackrwal, 2021-04-18:
I have a similar problem when using UART. 

```
from machine import UART

uart = UART(0, 9600)
uart.init(9600, bits=8, parity=None, stop=1)

  while True:
      uart.write(b'Hello World \r')
```

Upon running the above code I get;

```
WARNING:root:Unexpected echo. Expected b'%Run -c $EDITOR_CONTENT\r\n', got b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

However when attempting to stop and restart the code I get a blank line and nothing more as opposed to the usual 'help() for..'.  Interstingly pressing enter into the console from this point continued to produce unexpected echo warnings.

Restarting the device, and Thorny did not fix this and I had to re-flash my device to successfully connect Thorny again. (and proceeded to immediately run into the same problem).

Running Thonny on Win10.

biodives, 2021-05-18:
I ran into similar symptoms but apparently for a weird hardware-related reason. Perhaps this may help narrow down the issue. I'm using a Raspberry pico with micropython and thonny running on Windows 10. My project uses the Adafruit LPS25 pressure sensor communicating via I2C over gpio sda/scl pins 14/15 .
- If I power the LPS25 from the regular pico 3.3V power pin all works fine.

To allow pico to power up the LPS25 I connected the LPS25 Vin to GPIO16 (the LPS25 only uses tens of uA)
- If I power on the pico first and then connect the LPS25 to GPIO16 all works fine.
- If I power on the pico while the LPS25 is already connected to GPIO16 I get the following if I run my program

WARNING:root:Unexpected echo. Expected b'%Run -c $EDITOR_CONTENT\r\n', got b'buzz on\r\nbuzz on\r\nbuzz on'

- If I quit thonny, hook up the pico+LPS25, and then restart thonny I get a endless loop writing "buzz on"

buzz on
...
buzz on

The latter happens even if I start thonny without a loaded program so it really isn't running code, probably just trying to establish a serial link to the pico. If I restart with the LPS25 unpowered the problem disappears. If I have the LPS25 connected to gpio16 but disconnect the scl pin connection the problem disappears, but not if I only disconnect the sda connection. So is the scl pin causing some strange oscillation, even if not programmed to do anything. Weirder still, if I move I2C sda/scl to pins 10/11 the problem disappears and my program works as intended.

I guess the hardware issue is interesting but not the point of this topic.  I'm pretty new to microcontrollers so perhaps making some basic mistake but I hope it sheds some light on what may trigger the "buzz-on" "echo" messages from thonny.

aivarannamaa, 2021-11-17:
The problem was related to re-running the program without interrupting previous run first. Thonny was supposed to do it, but it didn't work out. 

The fix will be in 4.0. In the meanwhile the work-around is to interrupt the previous process with Ctrl+C before invoking the Run command

closed 2021-11-17 by aivarannamaa