thonny/thonny #1670: Inputting negative numbers crashes Micropython interpreter

evandavey, 2021-03-02:
```
>>> -1
[object_link_for_thonny=-1]-1
```

```
Traceback (most recent call last):
  File "C:\Users\evand\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1708, in event_generate
    handler(event)
  File "C:\Users\evand\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 433, in _handle_program_output
    self._update_visible_io(None)
  File "C:\Users\evand\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 512, in _update_visible_io
    self._apply_io_event(data, stream_name)
  File "C:\Users\evand\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 555, in _apply_io_event
    self.active_extra_tags.pop()
IndexError: pop from empty list
```
Tested on an ESP32.  Negative numbers seem to work OK in other interpreters.

aivarannamaa, 2021-03-02:
Thank you for the report!

I didn't expect that certain objects in MicroPython have negative id-s:

```
>>> print(id(-1))
-1
>>> print(id(-2))
-3
>>> print(id(-20))
-39
>>> print(id(20))
41
```

I commited the fix, but it will take some days until I have time for a new release. Until then please use `print` or apply the fix to your installation yourself:

* In Thonny select "Tools => Open Thonny program folder" and open shell.py
* Click on the commit link above and apply the same changes to your shell.py
* Restart Thonny.

closed 2021-03-06 by aivarannamaa

aivarannamaa, 2021-03-07:
@evandavey , the fix is now published in Thonny 3.3.6: https://github.com/thonny/thonny/releases/tag/v3.3.6