thonny/thonny #1674: ManagementError since Thonny 3.3.x

Crazyhackelkorn, 2021-03-03:
Hi,
i used Thonny 3.2.7 on macOS 11.2.2 with a Wio Terminal and an modified Ardupy. For Testing little code ideas.

Today i tried a newer Thony version 3.3.5, 3.3.5_alt, 3.3.2 and 3.3.0. All of these print a ManagementError, but the python code works fine

**Errormessage:**
[image]

**Python code:**
[image]
[image]

Now I tried running nothing this works but if I but anything insite the editor like space, enter it throws the error.

I found my Problem I have for bigger projekt two files /lib/os.py and /lib/uos.py for a device cross compatibility. Because ArduPy only has ```import os``` and most micropython uses ```import uos```

aivarannamaa, 2021-03-03:
Thank you for the report!

> Didn't found modul uos ...

I assume your code prints this. Please try what happens when you delete this print statement.

Crazyhackelkorn, 2021-03-03:
Yes, this print is from my code.
You were right. If I comment this print in ```/lib/uos.py``` out, my compatibility-file work as intended and no more ManagementError.

Thank you for your fast help.

closed 2021-03-03 by Crazyhackelkorn