thonny/thonny #1675: change from simple to regular mode reset size

ianken51, 2021-03-04:
I was using Thony 3.3.5 on Raspberry Pi 400 (Raspberry Pi OS) and tried "regular" mode in Thony.

It seemed OK, but when I reset it back to "simple" mode I now have an IDE that is almost unreadable. The font and the app shows miniscule text and icons. 

How does one go about resetting the app to default size?

aivarannamaa, 2021-03-04:
Have you tried the Zoom button?

Are you sure you didn't change "Tools => Options => UI scaling factor"  while in regular mode?

ianken51, 2021-03-04:
Thank you so much. I pretty much thought it would be my error, and it was.

Reset to “default” font and “simple” mode, and restarted Thony and it’s back as good as gold.

Thank you so much, Ian

ianken51, 2021-03-04:
On-line help solved what was my error. Must have accidentally set font to 0.5 and could only just read it. My bad. Thanks for help. I very much enjoy using Thony, so was annoyed that I messed it up. All good now 👌😀

closed 2021-03-04 by ianken51

aivarannamaa, 2021-03-04:
Great! Good luck!