thonny/thonny #1677: Issue installing most libraries

Yeetakus, 2021-03-04:
When I try to install Vector3D I get this error message
"
Installing to temp directory
Error installing 'vector3d': , packages may be partially installed
Installing to: C:\Users\Pascal\AppData\Local\Temp\tmpz784gvi0\
Warning: micropython.org SSL certificate is not validated

micropip returned with error code 1

"
I need this library for a mpu9125/50 module which will be used on the Pi Pico.

Any help will be greatly appreciated as I have been unable to do a lot of things because of this.
Thank you!

aivarannamaa, 2021-03-05:
Thank you for the report!

I assume you are using Thonny 3.3.3. Newest Thonny makes this situation more clear -- the SSL thing is just a warning and the real reason is that vector3d is not packaged for MicroPython. Actually, most of the libraries on PyPI can't be used with MicroPython.

When looking at the code of vector3d, it looks like it might actually work with MicroPython. You could try downloading it from PyPI or GitHub (https://github.com/kirill-belousov/vector3d/archive/v1.0.0.zip), extract the files and upload the `vector3d` directory to Pico's lib directory:

* Open Thonny's file browser (View => Files)
* Double click on `lib` on the lower pane
* Navigate to the location where you unpacked `vector3d` on the upper pane
* Right click on `vector3d` directory and select `Upload to lib`

closed 2021-03-09 by aivarannamaa

tyapspap, 2022-05-11:
Is this approach valid for Thonny 3.3.6? It's not working for me. When I open the file browser I don't see any 'lib'.

aivarannamaa, 2022-05-12:
>  When I open the file browser I don't see any 'lib'.

Your MicroPython or CircuitPython device may not have this library. In this case you may create it yourself.

tyapspap, 2022-05-13:
Thank you Sent from Mail for Windows From: Aivar AnnamaaSent: 13 May 2022 02:01To: thonny/thonnyCc: tyapspap; CommentSubject: Re: [thonny/thonny] Issue installing most libraries (#1677) When I open the file browser I don't see any 'lib'.Your MicroPython or CircuitPython device may not have this library. In this case you may create it yourself.