thonny/thonny #1679: numpy fails to install in thonny-3.3.6_alt.pkg on macOS Big Sur.

r0the, 2021-03-06:
Just out of curiosity, I tried out the alternative package with Python 3.9 on my macOS Big Sur.
When installing numpy through the package manager, I get an error (see below).

On my system Python 3.9.2, numpy 1.20.1 installs successfully.

On the «standard» Thonny 3.3.6, numpy 1.20.1 also installs successfully.

It's nothing that worries me, I just tought you'd like to know.

```
Collecting numpy
  Downloading numpy-1.20.1.zip (7.8 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/bin/python3.9 /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/tmp77y34k7s
         cwd: /private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy
    Complete output (237 lines):
    Running from numpy source directory.
    setup.py:485: UserWarning: Unrecognized setuptools command, proceeding with generating Cython sources and expanding templates
      run_build = parse_setuppy_commands()
    Cythonizing sources
    Processing numpy/random/_bounded_integers.pxd.in
    Processing numpy/random/_philox.pyx
    Processing numpy/random/_bounded_integers.pyx.in
    Processing numpy/random/_sfc64.pyx
    Processing numpy/random/_mt19937.pyx
    Processing numpy/random/bit_generator.pyx
    Processing numpy/random/mtrand.pyx
    Processing numpy/random/_generator.pyx
    Processing numpy/random/_pcg64.pyx
    Processing numpy/random/_common.pyx
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    blis_info:
      libraries blis not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    openblas_info:
      libraries openblas not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_3_10_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries tatlas not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_3_10_blas_info:
      libraries satlas not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries ptf77blas,ptcblas,atlas not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_blas_info:
      libraries f77blas,cblas,atlas not found in ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    /private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy/numpy/distutils/system_info.py:1989: UserWarning:
        Optimized (vendor) Blas libraries are not found.
[... 168 lines cut ...]
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-build-env-azr5dke5/overlay/lib/python3.9/site-packages/setuptools/command/dist_info.py", line 31, in run
        egg_info.run()
      File "/private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy/numpy/distutils/command/egg_info.py", line 24, in run
        self.run_command("build_src")
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy/numpy/distutils/command/build_src.py", line 144, in run
        self.build_sources()
      File "/private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy/numpy/distutils/command/build_src.py", line 155, in build_sources
        self.build_library_sources(*libname_info)
      File "/private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy/numpy/distutils/command/build_src.py", line 288, in build_library_sources
        sources = self.generate_sources(sources, (lib_name, build_info))
      File "/private/var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/pip-install-3ivs0k5u/numpy/numpy/distutils/command/build_src.py", line 378, in generate_sources
        source = func(extension, build_dir)
      File "numpy/core/setup.py", line 676, in get_mathlib_info
        raise RuntimeError("Broken toolchain: cannot link a simple C program")
    RuntimeError: Broken toolchain: cannot link a simple C program
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/bin/python3.9 /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /var/folders/gh/l6t4m6dj5rb0541622rnf6s00000gn/T/tmp77y34k7s Check the logs for full command output.
Process returned with code 1
```

aivarannamaa, 2021-03-07:
Thank you for the report!

This is good to know, as I considered publishing only the universal2 variant of Thonny+Python bundle since Thonny 4.0.
I read somewhere, that x64 packages should be usable with universal2 Python via Rosetta, but apparently it's not the case with Numpy.

Let's hope they publish universal2 wheels soon: https://github.com/numpy/numpy/issues/18143

aivarannamaa, 2022-04-01:
Seems to install nicely now in Thonny 4.0.0b1

closed 2022-04-01 by aivarannamaa