thonny/thonny #1680: Remote Python (SSH): File size error

Dano79sk, 2021-03-06:
Hello.
I am using Thonny with remote connection over ssh. The issue is, when size of my source file is over 3604 bytes, i get this error:
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
  File "/tmp/thonny-backend-3.3.5/thonny/plugins/cpython/__main__.py", line 43, in <module>
    MainCPythonBackend(target_cwd).mainloop()
  File "/tmp/thonny-backend-3.3.5/thonny/backend.py", line 62, in mainloop
    msg = self._fetch_next_incoming_message(timeout=0.01)
  File "/tmp/thonny-backend-3.3.5/thonny/plugins/cpython/cpython_backend.py", line 173, in _fetch_next_incoming_message
    self._read_one_incoming_message()
  File "/tmp/thonny-backend-3.3.5/thonny/backend.py", line 122, in _read_one_incoming_message
    msg = parse_message(line)
  File "/tmp/thonny-backend-3.3.5/thonny/common.py", line 228, in parse_message
    return eval(msg_string[1:])
  File "<string>", line 1
```

Thanks Dano

aivarannamaa, 2021-03-08:
Thank you for the report!

I could reproduce this, but I don't know why this happens. It looks like the command including the user code arrives to remote process intact, but it gets truncated when parsing with `eval`.

aivarannamaa, 2021-03-08:
I tried replacing `eval` with `ast.literal_eval`, but the result is the same -- 9K characters go in, but according to the exception, the parser sees only 4K first characters. It is strange, because when I'm doing the same with the same interpreter with simple subprocess (ie without SSH-ing to the same machine and interpreter), then everything works fine.

aivarannamaa, 2021-03-08:
I misinterpreted the problem. Actually it was about long lines being truncated in SSH communcation (Thonny used to pack your code into a single line before sending it to the back-end in order to simplify reading in commands).

I now refactored the code such that long commands get split into several lines. This almost solved your case, but occasionally (and quite rarely) there will be extra garbage printed to Thonny's Shell. This doesn't seem to affect the execution, though.

The fix will be in the first beta of Thonny 4.0.0. Until then the work-around is to save the script directly to the remote machine and let Thonny run it from there.