thonny/thonny #1681: Is it possible to clear console on every run?

pablodegrande, 2021-03-07:
I am planning to use Thonny to teach a group of kids how to code. I think the UI is excelent, compared to all too-complex stuff I have evaluated. The only feature I am missing is how to tell Thonny to clear the console everytime F5 is pressed (I am pretty sure that would create a much more clean experience for beginners, in terms of cause-consequence of every run). 

Thanks a lot, Pablo.

aivarannamaa, 2021-03-08:
Thank you for the suggestion!

The best I can do is to add an option for this, because the output from last runs may be occasionally interesting.

What do you think making the old output gray instead of deleting it?

pablodegrande, 2021-03-08:
I think deleting/clearing is better than graying. I understand it's a
console, but for the newbies, I think that the simpler, the better, and
clearing everything on every run avoids for the task of identifying actual
output vs previous output .

------------------

dmccreary, 2021-03-13:
Is there a way for a program running within the microcontroller to signal to the IDE to clear the terminal window?  That way the user writing the program could decide if the console should be cleared and add a ide.console.clear() function to the end of their program.

aivarannamaa, 2021-03-13:
@dmccreary, thank you for the idea! Providing a fake method for clearing would be hacky and difficult to implement, but I can implement support for standard ANSI code for clearing the screen. I now added this to my TODO list: #1697.

kintekobo, 2022-01-27:
Could the 'clear shell output before run' be made as an option perhaps? It is useful to have the error messages appear after a run when debugging but having the option to clear before running does help you to focus on the latest errors without being confused by previous output.

speedy-10, 2022-02-02:
Hi Pablodegrande, for me it's fine to use the shortcut 'crtl + L' to c**L**ear the shell. In some situations, esp. if I changed the code slightly, it's useful to compare the new output of the shell with the previous. So imho it should stay as it is :-) or be a further option.
Regards

pablodegrande, 2022-02-03:
Thonny is great to teach kids how to code. I've used to with 6 year old
kids. It main advantage in that scenario (compared to visual studio or any
other code editor i've seen) is it's simplicity. Kids can open the editor
at home, and only the code is shown, with a Play button in there. That's
fine.

However, if you ask me, tell them to press "ctrl+L" everytime time they run
code... well, it's not fine. And having the results of every run mixed
together, it's by no means reasonable if what you have to teach is that all
runs of a code are independent processes/executions. So, as a simple code
editor to learn/teach python, I guess the argument for clearing the console
is strong, both for practical and theoretical reasons.

aivarannamaa, 2022-02-04:
My current plan is to add the checkbox "Clear the Shell before running a script", which is checked by default. The change in the default behavior will be outlined in the release notes.

speedy-10, 2022-02-05:
Sounds good.

closed 2022-03-19 by aivarannamaa