thonny/thonny #1686: Thonny doesn't respond and window doesn't update

AdamNaghs, 2021-03-08:
Hi,
I'm using Thonny on an M1 Mac and have been loving Thonny, it's helped me understand so many concepts with the debug tool. I was about to use Thonny to get a better understanding of a for loop I wrote that does the same thing as the max() function until I pasted it into Thonny and since then Thonny has been completely broken. The window doesn't update, I can't edit anything, when I try to open another file nothing happens. I've tried uninstalling and re-downloading to no avail and I even went into my library folder and deleted the left over Thonny folder from the uninstall and that has not helped either. If anyone could tell me what I should do that would be greatly appreciated, Thonny has become something I use daily to understand concepts and I'd hate be forced to part ways with it.

aivarannamaa, 2021-03-08:
Thank you for the report!

Please go to `~/Library/Thonny` (ie. open `Library/Thonny` under your home directory) and find configuration.ini and send it to me. Also, there are a bunch of log-files, which may contain some clues.

You can post them separately or zip together and post here.

aivarannamaa, 2021-03-21:
@AdamNaghs, did you find a solution?

aivarannamaa, 2021-03-28:
Please re-open if you have more information

closed 2021-03-28 by aivarannamaa

tcp-22, 2023-02-13:
@aivarannamaa 
Hi, I have the same issue with Macbook Pro M1, the installation fails.

aivarannamaa, 2023-02-13:
@tcp-22, it looks like for @AdamNaghs the installation does not fail, so you don't have the same issue. Please open another issue and add more details about how the installation fails for you.