thonny/thonny #1687: Thonny not working

sentes68, 2021-03-08:
Traceback (most recent call last):
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1957, in toolbar_handler
    handler(*args)
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 471, in cmd_stop_restart
    self.restart_backend(True)
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 602, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 86, in __init__
    self._fix_port()
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 99, in _fix_port
    potential = self._detect_potential_ports()
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 191, in _detect_potential_ports
    all_ports = list_serial_ports()
  File "C:\Users\ProVideo\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 878, in list_serial_ports
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial.tools'

sentes68, 2021-03-08:
How reset Thonny
I try reinstall but still not work

aivarannamaa, 2021-03-08:
Thank you for the report!

Have you been able to use Thonny on this computer before?

Can you go to "Tools => Manage plug-ins"? Do you have `pyserial` in the left hand list? What about `serial`?

closed 2021-03-21 by aivarannamaa