thonny/thonny #1689: Not able to reload content of ESP32 with Micropython

jbelusic, 2021-03-08:
When ESP32 is connected to computer via data USB cabel,  if we push Stop/Restart backend button on Thonny IDE then content of ESP32 is not shown, UART is disconnected. Bug is that this can be connected and shown again if we do this steps:
1.  Right click on local file and then do send to ESP32
2. cancel operation
3. when Thonny ask that file will be rewritten then cancel again
Then, UART connects again and files are shown again.
I have Raspberry Pi Dark UI Theme and Sintax Theme Desert Sunset, everything else is default.

aivarannamaa, 2021-03-08:
Thank you for the report!

What happens when you close Thonny (while leaving ESP32 back-end selected) and launch it again? Do you get anything in the shell?

Do you have something in your main.py on the device?

Are you able to use the REPL via Thonny?

jbelusic, 2021-03-09:
Hi, I dont using REPL. I have logic code in main.py and this logic works
before new version (or maybe two versions down).
If I close Thonny and open it again, it does not reload content.

Jadranko Belušić
[image: https://www.linkedin.com/in/jadranko-belu%C5%A1i%C4%87-818231176/]
<https://www.linkedin.com/in/jadranko-belu%C5%A1i%C4%87-818231176/>

pon, 8. ožu 2021. u 22:06 Aivar Annamaa <notifications@github.com> napisao
je: