thonny/thonny #1691: Thonny and RP Pico Lock Up When Code Run Run From IDE

A-Krau, 2021-03-10:
When running the linked code, Thonny and RP Pico will freeze and require the RP Pico to be unplugged and plugged back in. After power cycle code will run fine.  Stopping and rerunning the same code with Thonny will cause the issue to return.

Commenting out the portion below in main.py will prevent the issue from happening, or at least delay when it happens.
`        if time.ticks_ms() - last_save >= save_int:
            print("Saving...")
            saved.write(int_dict)
            last_save = time.ticks_ms()`

Thonny 3.3.3
Windows 10 (64-bit)
Python 3.7.9 (32-bit)
Tk 8.6.9

pimoroni-pico-micropython.uf2 0.0.9
https://github.com/pimoroni/pimoroni-pico/releases

aivarannamaa, 2021-03-10:
Thank you for the report!

Have you tried commenting out only the print-statement at line 45?

A-Krau, 2021-03-10:
Good suggestion.

I just tried running it with link 45 commented out and it didn't seem to have an effect on the issue.  Commenting out lines 87 and 97 in Blinky.py also does not make a difference.

aivarannamaa, 2021-03-31:
@A-Krau, please check, whether the same problem occurs with newer firmware: https://github.com/pimoroni/pimoroni-pico/releases/tag/v0.1.0

aivarannamaa, 2021-11-17:
Closing, as Thonny and MicroPython have changed in the meanwhile

closed 2021-11-17 by aivarannamaa