thonny/thonny #1692: Thonny in Linux: can't set path for the virtual environment...

jonsafi, 2021-03-11:
Hello
edited  ~/.thonny/configuration.ini manually
and

closed 2021-03-11 by jonsafi

aivarannamaa, 2021-03-11:
In the e-mail I see that you mentioned version 2.1. This is not supported anymore. I suppose you have apt-installed it. I recommend apt-removing this and executing following command instead, which will download and install latest Thonny bundled with private Python 3.7:

```
bash <(wget -O - https://thonny.org/installer-for-linux)
```

jonsafi, 2021-03-11:
Many thanks for this tip! Had tried also installing Thonny via https://github.com/thonny/thonny/releases/tag/v3.3.6 but could not get the newer version. Am actually installing this for a group of students, and tried the command                                                              **wget -O - https://thonny.org/installer-for-linux) | sudo bash**                                                                                                                                              the command works, but seems it can't install it for all users (even if I do have permissions). Guess the command you kindly      gave should  work for students too, without requiring any  sudo privileges? Regards and thanks, Sami

aivarannamaa, 2021-03-11:
If you want to install to a shared location, then you need to download and unpack tar.gz for your platform and run `./install /your/target/location`

jonsafi, 2021-03-12:
Thanks so much, so just the very last part of installation changes for many users. I guess requiring each user or student to install it separately by herself/himsel using **(wget -O - https://thonny.org/installer-for-linux) | bash** might also work..?

aivarannamaa, 2021-03-12:
> I guess requiring each user or student to install it separately by herself/himsel using `(wget -O - https://thonny.org/installer-for-linux) | bash` might also work..?

Yes, if you don't have disk space problems and the students are not afraid of command line, then I would prefer this approach.

BTW. I like your command more so I replaced `bash <(wget -O - https://thonny.org/installer-for-linux)` with `(wget -O - https://thonny.org/installer-for-linux) | bash` on Thonny's homepage.

jonsafi, 2021-03-12:
Right, thanks, just lovely! Requiring students to install it by themselves will hopefully encourage them to use this great too! Glad to hear the pipeline made it to the Thonny homepage :-).