thonny/thonny #1693: Linux and Thonny: can't set the virtual environment

jonsafi, 2021-03-11:
Hello,
am using Thonny via Linux (v. 2.1.16) and it is working nicely.  Have set up a **virtual environment** 'venv_' at
~username/venv_/bin/python3, but can't get thonny to use it.
Tried it manually too, edited ~/.thonny/configuration.ini manually
and changed the backend configuration to: backend configuration =/~username/venv_/bin/python3, but when                                      I restart  thonny, get the following error msg:  __Interpreter at path. /~username/venv_/bin/python3 not found. Please
recheck corresponding option!__                                                                                                                                                                                              Am really puzzled, by this, any help would be greatly appreciated.                                                                                                   Thanks and, regards, Sami.

aivarannamaa, 2021-03-21:
Sorry for late reply!

`/~username/venv_/bin/python3` doesn't look like a valid path. Try omitting the first slash. Also, try writing the fullpath (`/home/.../...`)

Thonny 2.1 is really old. If your distro doesn't provide a newer version, then I recommend uninstalling it and installing fresh Thonny + Python bundle with:

```
bash <(wget -O - https://thonny.org/installer-for-linux)
```

You can still configure it to use your system Python (button on the lower-right corner of the window)

jonsafi, 2021-03-21:
Many thanks! Upgrading Thonny seems to have solved the issue :-)

closed 2021-03-21 by jonsafi