thonny/thonny #1695: Specify Interpreter / Connection at Startup?

thestumbler, 2021-03-13:
When using `Thonny` to connect to multiple boards, either over USB or WebREPL, how can you specify which connection at startup, or alternatively tell `Thonny` not to attempt to connect to anything?

For example, I am successfully connected to an ESP32 module in an instance of `Thonny`. I then start a second instance in order to connect to another  ESP32 module. This second instance will automatically try to connect to the first ESP32 and fail. I can then edit the interpreter / connection details and successfully connect to a second module. 

I don't think this is the intended workflow. Clearly I'm missing something.

If it helps, here are the error messages generated when instance 2 starts up and automatically tries to connect to the already-connected ESP32 module:

**Messages in instance 2:**
```
Traceback (most recent call last):
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\http.py", line 139, in read_response
    status_line = await read_line(stream)
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\http.py", line 219, in read_line
    raise EOFError("line without CRLF")
EOFError: line without CRLF

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 101, in read_http_response
    status_code, reason, headers = await read_response(self.reader)
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\http.py", line 141, in read_response
    raise EOFError("connection closed while reading HTTP status line") from exc
EOFError: connection closed while reading HTTP status line

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 73, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 547, in __await_impl__
    extra_headers=protocol.extra_headers,
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 290, in handshake
    status_code, response_headers = await self.read_http_response()
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 103, in read_http_response
    raise InvalidMessage("did not receive a valid HTTP response") from exc
websockets.exceptions.InvalidMessage: did not receive a valid HTTP response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\http.py", line 139, in read_response
    status_line = await read_line(stream)
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\http.py", line 219, in read_line
    raise EOFError("line without CRLF")
EOFError: line without CRLF

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 101, in read_http_response
    status_code, reason, headers = await read_response(self.reader)
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\http.py", line 141, in read_response
    raise EOFError("connection closed while reading HTTP status line") from exc
EOFError: connection closed while reading HTTP status line

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1580, in <module>
    connection = WebReplConnection(args["url"], args["password"])
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 47, in __init__
    raise res
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 60, in _ws_main
    await self._ws_connect()
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\webrepl_connection.py", line 76, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 547, in __await_impl__
    extra_headers=protocol.extra_headers,
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 290, in handshake
    status_code, response_headers = await self.read_http_response()
  File "C:\Users\rclott\AppData\Local\Programs\Thonny\lib\site-packages\websockets\client.py", line 103, in read_http_response
    raise InvalidMessage("did not receive a valid HTTP response") from exc
websockets.exceptions.InvalidMessage: did not receive a valid HTTP response

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
```
**Messages in instance 1:**
```

WebREPL connected
>>> 
MicroPython v1.14-22-gede6b86a0 on 2021-02-13; TTGO ESP32 with TTGO

Type "help()" for more information.
>>> 
Concurrent WebREPL connection from ('192.168.1.168', 58248) rejected

Concurrent WebREPL connection from ('192.168.1.168', 58249) rejected
```

aivarannamaa, 2021-03-13:
Thank you for the report!

I'll make sure the error message will be nicer (#1698), but I'm rather happy with the current approach of the second instance starting with the same preferences (including interpreter) as the first one.

I understand it's not perfect for your use case, but it's the simplest to implement and easiest to explain. Note that Thonny is (and will be) a beginner's IDE, therefore it favors simplicity over convenience by default.

thestumbler, 2021-03-13:
Regarding the default behavior, I completely agree with you. I also want it to use the previous connection by default. 

The point of my inquiry was to ask if there is any command line argument we could give to either (a) specify a connection, or (b) don’t connect at all? For example 

`$ thonny --noconnect`

or 

`$ thonny --connect ESP32`

aivarannamaa, 2021-03-15:
No, there is no such feature. 

I'm considering implementing a more general solution (possibility to override any configuration value from the command line, #1703), which hopefully is acceptable for your case as well.

closed 2021-03-15 by aivarannamaa