thonny/thonny #1696: Installation 3.3.6 does not overwrite older version

speedy-10, 2021-03-13:
Hi, until now a new installation did overwrite the old one, but now I'm not asked (Y/n), it is cancled at once.
Seen on Ubuntu 20.04.
Regards
terminal:
`~$ (wget -O - https://thonny.org/installer-for-linux) | bash
--2021-03-13 11:23:41--  https://thonny.org/installer-for-linux
Auflösen des Hostnamens thonny.org (thonny.org) … 185.7.252.74
Verbindungsaufbau zu thonny.org (thonny.org)|185.7.252.74|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 762
Wird in »STDOUT« gespeichert.

-                   100%[===================>]     762  --.-KB/s    in 0s      

2021-03-13 11:23:41 (72,7 MB/s) - auf die Standardausgabe geschrieben [762/762]

This script will download and install Thonny (thonny-3.3.6) for Linux (32 or 64-bit PC).
Downloading https://github.com/thonny/thonny/releases/download/v3.3.6/thonny-3.3.6-x86_64.tar.gz
--2021-03-13 11:23:41--  https://github.com/thonny/thonny/releases/download/v3.3.6/thonny-3.3.6-x86_64.tar.gz
Auflösen des Hostnamens github.com (github.com) … 140.82.121.4
Verbindungsaufbau zu github.com (github.com)|140.82.121.4|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://github-releases.githubusercontent.com/163728962/4bad3e00-7e8f-11eb-8f33-74eba43fdbe5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210313%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210313T102342Z&X-Amz-Expires=300&X-Amz-Signature=0b80be2f018903e57fcd1c935b7533012ecfda3270ba348efed30f9fb8337063&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=163728962&response-content-disposition=attachment%3B%20filename%3Dthonny-3.3.6-x86_64.tar.gz&response-content-type=application%2Foctet-stream [folgend]
--2021-03-13 11:23:42--  https://github-releases.githubusercontent.com/163728962/4bad3e00-7e8f-11eb-8f33-74eba43fdbe5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210313%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210313T102342Z&X-Amz-Expires=300&X-Amz-Signature=0b80be2f018903e57fcd1c935b7533012ecfda3270ba348efed30f9fb8337063&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=163728962&response-content-disposition=attachment%3B%20filename%3Dthonny-3.3.6-x86_64.tar.gz&response-content-type=application%2Foctet-stream
Auflösen des Hostnamens github-releases.githubusercontent.com (github-releases.githubusercontent.com) … 185.199.109.154, 185.199.110.154, 185.199.111.154, ...
Verbindungsaufbau zu github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.109.154|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 33270232 (32M) [application/octet-stream]
Wird in »./tmp.hStzO2QNDQ/thonny-3.3.6-x86_64.tar.gz« gespeichert.

./tmp.hStzO2QNDQ/thonn 100%[===========================>]  31,73M  1,38MB/s    in 23s     

2021-03-13 11:24:05 (1,38 MB/s) - »./tmp.hStzO2QNDQ/thonny-3.3.6-x86_64.tar.gz« gespeichert [33270232/33270232]

Copying files to /home/USER/apps/thonny .......................... 
/home/USER/apps/thonny already exists. I need to clear it. Is it OK? [Y/n]: Installation is cancelled
`

aivarannamaa, 2021-03-13:
Thank you for the report!

I recently updated the installation command on Thonny's webpage from `bash <(wget -O - https://thonny.org/installer-for-linux)` to `(wget -O - https://thonny.org/installer-for-linux) | bash`, without properly testing it. 

I now reverted the recommended command to `bash <(wget -O - https://thonny.org/installer-for-linux)`.

@jonsafi (#1692), it looks like the pipeline does not work properly with user input.

jonsafi, 2021-03-13:
Right, thanks for this Aivar, did not realize this inability to overwrite was due to the pipeline. Good thing indeed that you fixed it back.

closed 2021-03-13 by aivarannamaa

speedy-10, 2021-03-13:
Fixed, thank you