thonny/thonny #1700: Can't upgrade Thonny on Raspberry Pi OS

dmccreary, 2021-03-13:
Hello,

I am trying to teach kids Python using the new Raspberry Pi Pico and Thonny.  I am using a Raspberry Pi 4 to run our classes.

I can't seem to upgrade Thonny on my Raspberry Pi OS.  The version that is installed is the old version 3.2.0.  When I try to use the command:

 sudo apt-get install python3-thonny

I get errors:
dpkg: warning: files list file for package 'python-rpi.gpio' missing; assuming package has no files currently installed
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'bubblewrap' contains empty filename
E: Sub-process /usr/bin/dpkg returned an error code (2)

pip3 install thonny --upgrade
That works but the app does not reflect the right version:

$ pip3 freeze | grep thonny
thonny==3.3.6

I also ran the dpkg list command:

$ dpkg -l | grep thonny
ii  thonny                                3.2.0-1+rpt2                        all          Python IDE for beginners

I also tried the shell script:
./thonny-3.3.6.bash 
Thonny only provides pre-built bundles for i686 and x86_64 (not armv7l)

Is there a way that we can install it on the Raspberry Pi OS using armv7l

Thank you for your help.

dmccreary, 2021-03-13:
I found a hack workaround.  I just went in and edited the /usr/bin/thonny file, which is just a small python file.  I changed all the references to the version number (three places, one in a comment) from the old 3.2 version to the new 3.3.6 version and everything worked great!!

I think this has to do with a bug in the package management tools, not the thonny source.  However, a "upgrade" menu in the GUI application would be really nice for future projects!

Thank you again for your great tool.

I also put a copy of the new file in /usr/bin/thonny below:

#!/usr/bin/python3
# EASY-INSTALL-ENTRY-SCRIPT: 'thonny==3.3.6','gui_scripts','thonny'
__requires__ = 'thonny==3.3.6'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('thonny==3.3.6', 'gui_scripts', 'thonny')()
    )

aivarannamaa, 2021-03-13:
Thank you for the report!

First, `python3-thonny` is deprecated, it's now `thonny` on all Debian based systems.

Regarding versions mismatch -- can it be that you have done `pip install --user thonny` (or another package which has `thonny` as its dependency)? This way you may end up with two versions of Thonny, with the one under user site packages taking precedence over the apt-installed one. Starting with pip version 20 this can happen even without `--user` switch.

So I recommend you `apt remove thonny` and `pip3 uninstall thonny` and then `apt install thonny` again.

Ivoz, 2021-03-16:
Yes, if you have a *system* tool present (one that is found in global system files, like `/usr/bin`) you do **not** want to try to manage it multiple ways. In this case multiple ways would be trying to upgrade it with both `apt` and a global `pip` command. You should choose either one or the other.

Nowadays mostly pip should not *overwrite* system placed files but you will still be very confused if you try both ways and don't know the ins and outs of who and how which binaries get priority in loading and on `PATH`s. 

AFAIK RPi OS itself updated to a very recent version of Thonny 3.3 to make the RPi Pico work, so a very recent version of thonny *should* already be in the apt repositories. If that is not recent enough for you, uninstalling and doing a "user install" with pip (by prepending `--user` to install command, see [here](https://pip.pypa.io/en/stable/user_guide/#user-installs) to update your path for this).

As Aivarannamaa said usually a scorched-earth uninstall method (wipe everything possibly to do with Thonny[ies]) followed by reinstalling one way is easiest to resolve this.

closed 2021-03-21 by aivarannamaa