thonny/thonny #1701: Thonny package install error - can not combine '--user

dandavies80, 2021-03-13:
When I try to install certain packages in Thonny, I get the following error

-----------------------------------------------
ERROR: Can not combine '--user' and '--target'

Process failed, return code: 1
-----------------------------------------------

This happens when I try to install PyGame, and Tweepy, but it works for most other modules.  Can anyone tell me what's going on and how I can fix this?

Thank you!

aivarannamaa, 2021-03-13:
Thank you for the report!

Which platform and Thonny version are you using? 

I can't reproduce this with Thonny 3.3.6 on Ubuntu 20.04 and Pygame 2.0.1

dandavies80, 2021-03-14:
Hi, thank you for replying.

I'm using Thonny 3.3.6 on Mac OS Catalina v 10.15.7.

Were you able to reproduce it with Tweepy at all?

Thanks for your help.

Dan

dandavies80, 2021-03-15:
> Thank you for the report!
> 
> Which platform and Thonny version are you using?
> 
> I can't reproduce this with Thonny 3.3.6 on Ubuntu 20.04 and Pygame 2.0.1

dandavies80, 2021-03-15:
Any ideas?  Thanks!

aivarannamaa, 2021-03-15:
Strange -- I could install both pygame and tweepy on Catalina with Thonny 3.3.6 without problems.

Are you using "Tools => Manage packages"? 

Are you using the default interpreter, ie. the one bundled with Thonny?

dandavies80, 2021-03-15:
Yes, I'm using "Tools => Manage packages", and using the default interpreter, the same one that runs Thonny.  I wonder what's different about my setup.

aivarannamaa, 2021-03-21:
It may be either a `PIP_*` environment variable or a global pip configuration file.

You can investigate environment variables like this:

```
import os
for key in os.environ:
    print("*", key, "=", os.environ[key])
```

See https://pip.pypa.io/en/stable/user_guide/#config-file for possible locations of pip config file.

aivarannamaa, 2021-03-28:
Did you find a solution?

dandavies80, 2021-03-29:
Hi Aivar,

No, I wasn't able to fix this issue.  Do you have any other ideas?

Dan

aivarannamaa, 2021-03-29:
Does it mean you don't have any `PIP_*` environment variables nor pip configuration files?

dandavies80, 2021-03-30:
Actually, I can't find those!  Do you know where they are on a Mac?  I'm
using Catalina v 10.15.7.

Dan

dandavies80, 2021-03-30:
If I do, I don’t know where they are.  Do you know where I can look for them?

aivarannamaa, 2021-04-09:
> If I do, I don’t know where they are. Do you know where I can look for them?

Please see the comment where I gave the snippet for examining environment variables and link about pip config files.

aivarannamaa, 2021-04-20:
Please reopen if you're still stuck or you have more details to add!

closed 2021-04-20 by aivarannamaa