thonny/thonny #1702: THONNY FAILED TO EXECUTE COMMAND get_globals

RobertoZaffa, 2021-03-14:
I'm using  Thonny veriosn 3.3.6 and I'm not sure that LEGO Hub is supported.
As soon the PC is connected to the hub, pressing enter in the shell, Thonny show the error message.
[image]
I hope this problem can be fixed. I would use thonny instead of LEGO App. Thank you very much

aivarannamaa, 2021-03-14:
Thank you for the report

> As soon the PC is connected to the hub, pressing enter in the shell, Thonny show the error message.

On the screenshot it looks like you had run your program and your program has an error. 

Do you get the same error when you fix your program and run it again?

What about evaluating simple expressions in the Shell?

aivarannamaa, 2021-03-28:
Closing as I can't progress without more information. Please reopen if you have more information.

RobertoZaffa, 2021-03-29:
Hi Aivar,
I tried to connect to the Hub without any programs. Everything seems ok
using simple commands but when I try to access the hub file system, the
error appears that I report to you in the attached files.
I also tried using the mindstorm library import commands but these don't
find the modules. Is it correct that with Thonny I cannot use the Lego
Mindstorm libraries?
Thanks a lot for your help

aivarannamaa, 2021-03-29:
> when I try to access the hub file system, the error appears that I report to you in the attached files.

Do you mean opening hub's files in Thonny? Or just doing `open(file)` in your programs?

> Is it correct that with Thonny I cannot use the Lego Mindstorm libraries?

You should be able to import the same modules that you can import in LEGO's Python editor. Can you give me an example import, which fails in Thonny?

aivarannamaa, 2021-05-01:
@RobertoZaffa, did you connect via USB cable or Bluetooth?

closed 2021-06-13 by aivarannamaa