thonny/thonny #1705: First-run wizard to choose a dark theme?

Ivoz, 2021-03-16:
Hi, I am thinking about the adoption marketing of thonny. I think these days most people (particularly younger generation) have forgot any association that a white theme is a starter, easy to read (possibly consider it the opposite!), "default" or "status quo". I don't have any numbers but a number of personal interactions that these days people will instantly prefer a dark themed app or website. There's even a small % of netizens who actively run from anything that can't be switched to a dark theme nowadays (OK, I'm slightly biased, I kinda fall in that group too).

So nowadays if a beginner hasn't yet been shown a new python-editor to use and really dig in with... I'm thinking that often simply seeing that VS Code comes in a cool dark theme and Thonny is some old rustic white thing can literally make the decision for them. They see the one that looks _cooler_ and pick that. Yes, after some clicks in the options you can totally reverse that in Thonny; but as always, first impressions count the most. 

Both [VS Code](https://code.visualstudio.com/) and [Visual Studio](https://visualstudio.microsoft.com/vs/) advertises themselves on their homepage with a dark theme; even [PyCharm](https://www.jetbrains.com/pycharm/) does the same now too.

So I propose either making dark theme default for new Thonny installation, or if that is not to your liking, possibly a simple first run wizard asking the user whether they'd like a dark or light configuration (I currently use Clean Dark UI & Tomorrow Night; I like the darker background that Tomorrow Night gives over Default Dark, personally).

It would also be possible to pick up a Light / Dark starting default from a current OS preference but I'd completely understand if you don't feel like writing all the OS-specific code to detect that ATM.

A screenshot in the Thonny website of Thonny looking dark might also help, to show a user that visits there, that the editor is not only possible in white. I was thinking maybe I could help with this last part at the least. Was the website built with any particular tools?

aivarannamaa, 2021-03-16:
Thank you for the suggestions!

Actually, I considered adding light/dark choice to the initial welcome screen. Regarding OS preference -- recent Tk version allows querying it in macOS. Things are less clear in Windows 10 (a related issue: #1664) and I haven't started researching Linux yet.

Regarding defaults -- currently I'd prefer using the current OS setting both for Thonny's welcome screen and homepage. I'd be happy if you could help with the homepage. It is all hand-written.

Ivoz, 2022-07-08:
Would it be possible to have this in the installer? Like if you select an option during install it creates user config with dark theme for UI & editor

aivarannamaa, 2022-07-08:
Having it in installer would be inconvenient for several reasons. First -- it would require different approach for different installers (Win, Mac, Linux). Second -- it is not obvious how to initialize user setting when running Win installer in admin mode and installing for all users.