thonny/thonny #1710: Ctrl+C not recognized in the shell

mristin, 2021-03-17:
Hi,
While developing [thonny-crosshair], we noticed that the watch process running in the shell (`Check the current file with CrossHair continuously`) can not be stopped with `ctrl+c`. The stop button in the toolbar works, though.

[thonny-crosshair]: https://github.com/mristin/thonny-crosshair/

I am running Thonny on Windows.

Is this by design (*e.g.*, so that user is supposed to always use the stop button)? Or do we need to fix something on our side? I searched the issues, but could not find anything related.

Thanks in advance for you response!

aivarannamaa, 2021-03-17:
Thank you for the report!

I'd call this a bug. I'll investigate later, as I'm afraid the fix isn't trivial. Thonny already contains some hacks to support interrupts in Windows without requiring a console window, but it looks like something more is required for interrupting subprocesses.

The stop button works, because it's simply killing the back-end process together with all child processes.

qouoq, 2023-01-12:
Would it be a reasonable approach to intercept Ctrl+C and trigger that same back-end process killing, when on Windows?

The current behavior has been the most disconcerting so far for me with Thonny. I haven't looked at more in detail, I only saw that a break on an input command doesn't respond before <enter>, and same for an infinite loop.

@aivarannamaa, where should one look to get an idea about the current hacks for Windows interrupts support?

aivarannamaa, 2023-01-13:
@qouoq, simply Ctrl-C with as kill doesn't feel right.

Here is the code that gets run when Ctrl-C gets detected (`self._proc` is the subprocess): https://github.com/thonny/thonny/blob/b857046a563dfae50d46a6fa8ff37546cc02f67f/thonny/plugins/cpython_frontend/cp_front.py#L105

What may affect the effect of this code, is here: https://github.com/thonny/thonny/blob/b857046a563dfae50d46a6fa8ff37546cc02f67f/thonny/running.py#L710

Please note that Ctrl-C will be interpreted as Copy, if a selected text is detected. I hope this detection works nicely, but in cases you can't interrupt with Ctrl-C, it's worth trying "Run => Interrupt execution" as well.

aivarannamaa, 2023-01-13:
I once also experimented with interrupting the backend process from within the backend process itself, by having the GUI send a respective message to the backend using approach similar to https://superfastpython.com/interrupt-the-main-thread-in-python/, but it broke in some situation, can't remember anymore. Maybe I should revisit this idea, then I could get rid of this console allocation thing as well ...

aivarannamaa, 2023-01-13:
I now found the issue which blocked the approach I just described: #1363