thonny/thonny #1712: Unable to run 'kivy.app' on Thonny

smfrederico, 2021-03-17:
I tried to run this code below:
from kivy.app import App
from kivy.uix.screenmanager import ScreenManager, Screen
from kivy.lang import Builder
Builder.load_file('frontend.kv')
class FirstScreen(Screen):
    def search_image(self):
        pass
class RootWidget(ScreenManager):
    pass
class MainApp(App):
    def build(self):
        return RootWidget()
    
MainApp().run()

But it failed. I got the following message:
[INFO   ] [Image       ] Providers: img_tex, img_dds (img_sdl2, img_pil, img_ffpyplayer ignored)
[CRITICAL] [Window      ] Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
sdl2 - ImportError: DLL load failed: Não foi possível encontrar o módulo especificado.
  File "C:\Users\Frederico\AppData\Roaming\Python\Python37\site-packages\kivy\core\__init__.py", line 62, in core_select_lib
    fromlist=[modulename], level=0)
  File "C:\Users\Frederico\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 314, in _custom_import
    module = self._original_import(*args, **kw)
  File "C:\Users\Frederico\AppData\Roaming\Python\Python37\site-packages\kivy\core\window\window_sdl2.py", line 27, in <module>
    from kivy.core.window._window_sdl2 import _WindowSDL2Storage
  File "C:\Users\Frederico\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython\cpython_backend.py", line 314, in _custom_import
    module = self._original_import(*args, **kw)

[CRITICAL] [App         ] Unable to get a Window, abort.

Could you enlighten me in this matter? I believe I wrote the code correctly, I wonder why I was unable to open the window.

smfrederico, 2021-03-17:
The version of Thonny is 3.3.6 / Windows 10/ Python 3.7.9/ Tk 8.6.9

aivarannamaa, 2021-03-18:
How did you install kivy?
When I installed kivy with Thonny's package manager, then I could run your code up to ` No such file or directory: 'frontend.kv'`

smfrederico, 2021-03-18:
Hello and Good Morining, Mr. Aivar Annamaa !

Very glad to hear from you! Could you please rewrite the lines below as
follows?

from kivy.app import App
from kivy.uix.screenmanager import ScreenManager, Screen
# from kivy.lang import Builder
# Builder.load_file('frontend.kv')
class FirstScreen(Screen):
    def search_image(self):
        pass
class RootWidget(ScreenManager):
    pass
class MainApp(App):
    def build(self):
        return RootWidget()

MainApp().run()

This "frontend" archive should be placed in the same directory as the
main.py. But this archive is not needed to open the app window.
Unfortunately, although small, I don´t have a copy of frontend right
now. Please, if you manage to open the window app, could you tell me
why? Thank you a lot indeed!

smfrederico, 2021-03-18:
Professor Ardit Sulce, owner of site PythonHow and instrictor at Udemy, solved the problem:
"Hi Frederico, I suspect that might be happening because Kivy has not been fully installed for your Python interpreter. This might be because Thonny uses Python 3.7 and in PyCharm in the videos, I was using a newer version of Python.

You could try to execute the following commands in the Thonny Python shell. They are supposed to install kivy and a few kivy dependencies.

!pip3.7 install kivy

!pip3.7 install kivy.deps.glew

!pip3.7 install docutils pygments pypiwin32 kivy.deps.sdl2

Please note that the commands assume your Thonny is using Python 3.7."  It worked perfectly. Thanks a lot Professor Aivar Annamaa !

closed 2021-03-18 by smfrederico