thonny/thonny #1713: Error on right clicking in local file explorer

eugeniobaglieri, 2021-03-18:
Each time that I right click on the local file section this error appears
```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/base_file_browser.py", line 591, in on_secondary_click
    self.refresh_menu(context="item")
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/base_file_browser.py", line 605, in refresh_menu
    self.add_middle_menu_items(context)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/files.py", line 219, in add_middle_menu_items
    self.check_add_upload_command()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/files.py", line 216, in check_add_upload_command
    self.menu.add_command(label=tr("Upload to %s") % target_dir_desc, command=_upload)
ValueError: incomplete format
```
[image]

aivarannamaa, 2021-03-18:
Thank you for the report!

I was not able to reproduce this. From your screenshot it looks like the device hasn't been able to list the files on the board's filesystem.  Did you get REPL prompt (`>>> `) in the Shell or is your device currently running your code?

eugeniobaglieri, 2021-03-18:
i can get the REPL on the connected device, hower the appears even with no device connected, simply clicking with right mouse button on local file explorer area.
I tried also on my notebook with big sur

eugeniobaglieri, 2021-03-20:
if it can be useful... I resolved modifying the line 216 of file
"/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/files.py" as indicated by the error.
Turning this
```
self.menu.add_command(label=tr("Upload to %s") % target_dir_desc, command=_upload)
```
into this
```
self.menu.add_command(label=tr("Upload to {}").format(target_dir_desc), command=_upload)
```

aivarannamaa, 2021-03-21:
@eugeniobaglieri, how does the "Upload to ..." command read in your context menu after your fix?

aivarannamaa, 2021-03-21:
Nevermind, I found the reason -- Italian translation was broken for this term.

closed 2021-03-21 by aivarannamaa

aivarannamaa, 2021-04-30:
Please try the new release: https://github.com/thonny/thonny/releases/tag/v3.3.7