thonny/thonny #1714: Thonny crashes on saving a file

althedoc, 2021-03-18:
I open Thonny (Fedora 33), click "Save as". I use any folder to save in and Thonny crashes:

Traceback (most recent call last):
  File "/usr/lib64/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/usr/lib/python3.9/site-packages/thonny/running.py", line 382, in cmd_run_current_script
    self.execute_current("Run")
  File "/usr/lib/python3.9/site-packages/thonny/running.py", line 337, in execute_current
    filename = get_saved_current_script_filename()
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 957, in get_saved_current_script_filename
    filename = editor.get_filename(force)
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 72, in get_filename
    self.save_file()
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 234, in save_file
    result = self.write_local_file(save_filename, content_bytes, save_copy)
  File "/usr/lib/python3.9/site-packages/thonny/code.py", line 247, in write_local_file
    f = open(save_filename, mode="wb")
FileNotFoundError: [Errno 2] No such file or directory: 'WINDOW DECORATIONS RELOADED\n/home/alex/Desktop/a.py'

aivarannamaa, 2021-03-18:
Thank you for the report!

This seems to be known issue with `zenity` utility. Please see #1550 for explanation and a work-around.

althedoc, 2021-03-18:
Thanks - that has solved it :)
It must be a fedora issue. I uninstalled (dnf) thonny and did pip3 install thonny - and the tk tickbox mentioned.

closed 2021-03-18 by althedoc