thonny/thonny #1718: asm_pio error when working in with RP-pico in shell mode

BarbabennoHBe, 2021-03-21:
I am trying the example in the 'Get started with MicroPython on Raspberry Pi Pico' book.
When I create a program in a file, it seems to work. But if I start it using the shell, I get an error.

Source:
```
@asm_pio(set_init=PIO.OUT_LOW)
def led_quarter_brightness():
    set(pins, 0) [2]
    set(pins, 1)
```

Error message
```
Traceback (most recent call last):
File "<stdin>", line 5, in <module>
File "rp2.py", line 254, in dec
File "<stdin>", line 7, in led_quarter_brightness
NameError: name '__thonny_helper' isn't defined
```

The software I used:
- MicroPython v1.14 on 2021-03-19; Raspberry Pi Pico with RP2040
- Thonny 3.3.6
- Ubuntu 20.04
- Python 3.8.5

Would it be possible to work with the @asm_pio when using the Shell?

aivarannamaa, 2021-03-21:
Thank you for the report!

>  if I start it using the shell

Do you mean running the whole code in the Shell? By using Thonny's "Run" command or by pasting/entering the whole code to the Shell?

Can you tell me more precisely how to reproduce this problem?

BarbabennoHBe, 2021-03-22:
I do the following:
1. Start Thonny
2. Disable the current program which might be running on the pico
3. Pasting the program above in the Shell box (after the >>>) and press enter. (When trying this now, I realised that I forgot to add the import in the example above, so I have just add it now).

When I place the program in a file on the pico, the program runs fine.
Also when I place other programs in the Shell part, after the >>>, this seems to work well.
So it seems to be a combination of the Shell and asm_pico (and Thonny and the pico of course)

I hope this helps

aivarannamaa, 2021-03-23:
Thank you! I could reproduce and fix it now. The fix will be in the first beta of Thonny 4.0

closed 2021-03-23 by aivarannamaa