thonny/thonny #1726: raspberry pie for dummies ,connecting to mcpi in python for minecraft

johnrogersavs, 2021-03-23:
I have received following back
   raise RequestError("%s failed"%self.lastSent.strip())
mcpi.connection.RequestError: world.getBlock(8,1,False) failed
any help appreciated
commented program attached from raspberry pie for dummies

 code in program attached

import sys, random  
from mcpi import minecraft
mc = minecraft.Minecraft.create()
mc.postToChat("Welcome to Minecraft Maze!")

def realx(x):
    return MAZE_X + (x*2) -1
 
def realz(z):
    return MAZE_Z + (z*2) -1

def showMaker(x, z):
    mc.setBlock(realx(x), GROUND+1, realz(z), 41) # 41=gold

def hideMaker(x, z):
    mc.setBlock(realx(x), GROUND+1, realz(z),0)
    
def demolish(realx, realz):
    mc.setBlocks(realx, GROUND+1, realz, realx,
        HEIGHT+GROUND, realz, 0)
   
def testAllWalls(cellx, cellz):
    if mc.getBlock(realx(cellx)+1, GROUND+1,
      realz(cellz)==MAZE_MATERIAL and mc.getBlock
      (realx(cellx)-1, GROUND+1, realz(cellz))==MAZE_MATERIAL
      and mc.getBlock(realx(cellx), GROUND+1, realz(cellz))+1==
      MAZE_MATERIAL and mc.getBlock(realx(cellx), GROUND+1,
      realz(cellz)-1))==MAZE_MATERIAL:
        return True
    else:
        return False
       
        
    
mc.setting("world_immutable", True)

#set maze size
SIZE = 10
HEIGHT = 2
MAZE_X = 0
GROUND = 0
MAZE_Z = 0
MAZE_MATERIAL = 1 #stone
GROUND_MATERIAL = 2 #grass
CEILING = False

# clear area
mc.setBlocks(MAZE_X-10, GROUND, MAZE_Z-10,
   MAZE_X+(SIZE*2)+10, GROUND+150, MAZE_Z+(SIZE*2)+10, 0)
#lay ground
mc.setBlocks(MAZE_X-10, GROUND, MAZE_Z-10,
    MAZE_X+(SIZE*2)+10, GROUND+150, MAZE_Z+(SIZE*2)+10,
    GROUND_MATERIAL)
mc.setBlock(MAZE_X, GROUND+HEIGHT+1, MAZE_Z,
  MAZE_MATERIAL)

mc.player.setTilePos(0, 100, 0)

#build wall grid
for line in range(0, (SIZE+1)*2, 2):
    mc.setBlocks(MAZE_X+line, GROUND+1, MAZE_Z,
      MAZE_X+line, GROUND+HEIGHT, MAZE_Z+(SIZE*2),
      MAZE_MATERIAL)
    mc.setBlocks(MAZE_X, GROUND+1, MAZE_Z+line,
      MAZE_X+(SIZE*2), GROUND+HEIGHT, MAZE_Z+line, MAZE_MATERIAL)
    
#set up variables for creating maze
numberOfCells = SIZE * SIZE
numberOfVisitedCells = 1  # 1 denotes the one you start in
cellsVisitedList = []
 
xposition = random.randint(1, SIZE)
zposition = random.randint(1, SIZE)
playerx = xposition
playerz = zposition
showMaker(xposition, zposition)
cellsVisitedList.append((xposition,zposition))

while numberOfVisitedCells < numberOfCells:
    possibleDirections = []
    
    if testAllWalls(xposition -1, zposition):
        possibleDirections.append("left")
        
    if testAllWalls(xposition +1, zposition):
        possibleDirections.append("right")
        
    if testAllWalls(xposition, zposition -1):
        possibleDirections.append("up")
        
    if testAllWalls(xposition, zposition +1):
        possibleDirections.append("down")
        
    hideMaker(xposition, zposition)
 
if len(possibleDirections) !=0:
    directionChosen=random.choice(possibleDirections)
        
        #knock down wall in direction chosen
    if directionChosen == "left":
        demolish(realx(xposition) - 1, realz(zposition))
        xposition -= 1
        
    if directionChosen == "right":
        demolish(realx(xposition) + 1, realz(zposition))
        xposition += 1
        
    if directionChosen == "up":
        demolish(realx(xposition), realz(zposition) - 1)
        zposition -= 1
        
        
    if directionChosen == "down":
        demolish(realx(xposition), realz(zposition) + 1)
        zposition -= 1
        
    numberOfVisitedCells += 1
        
    #after the move, increase number of visited cells
        
    cellsVisitedList.append((xposition, zposition))
    showMaker(xposition, zposition)
        
else: 
    retrace = cellsVisitedList.pop()
    xposition = retrace[0]
    zposition = retrace[1]
    
    showMaker(xposition, zposition)
       
if CEILING == True:
    mc.setBlocks(MAZE_X, GROUND+HEIGHT+1, MAZE_Z,
      MAZE_X+(SIZE*2), GROUND+HEIGHT+1, MAZE_Z+(SIZE*2), 20)
mc.postToChat("Your maze is ready!")
mc.postToChat("Happy exploring!")
mc.player.setTilepos(realx(playerx), GROUND+1,
  realz(playerz))

aivarannamaa, 2021-03-23:
Do you think it's Thonny's problem?

Can you run your code outside of Thonny?

johnrogersavs, 2021-03-24:
I have only run it in Thonny.
I believe it may be a problem with the block building code  or having
access to the module that allows you to build the blocks.

Running the following very simple program.
import sys, random
from mcpi import  minecraft
mc = minecraft.MinecraftCreate()
mc.post ToChat("welcome to Minecraft Maze!")    #brings up welcome statement
mcplayer.setTilepos(0, 100, 0) # parachutes down into centre of screen

# all runs ok
then the following should build a cube , but nothing happens

#note  the below code should demonstrate the block building function

mc.setBlocks(0, 0, 0, 10, 5, 7, 45)    the number 45 denotes brick
mc.setBlocks(1, 0, 1, 9, 5, 6, 0)       the number 0 denotes air
This should build a cube 10*7*5 ,  1 brick thick and filled with air

the full program i have de bugged so there are no syntax errors or
unbalanced brackets or unexpected indent errors, thus i feel it is down to
either a code error or the inability to communicate with block
building block  function

aivarannamaa, 2021-03-24:
If you can run some Minecraft programs, then I believe it's not Thonny's problem.

I recommend asking for help at this forum: https://www.raspberrypi.org/forums/viewforum.php?f=32

closed 2021-03-24 by aivarannamaa