thonny/thonny #1727: micropython library

spacehz, 2021-03-24:
hi, i wanted to install blynk library (for my ESP8266) using tools and manage packages but an an error occurs saying that "micropip returned with error code 1"  despite it's a micropython library !

aivarannamaa, 2021-03-24:
Thank you for the report!

I assume you are referring to `blynklib`. It looks like this is meant to be installed with pip, not with upip or micropip, which Thonny is using with MicroPython back-end.

As far as I understand, you need to download https://raw.githubusercontent.com/blynkkk/lib-python/master/blynklib_mp.py (and optionally https://raw.githubusercontent.com/blynkkk/lib-python/master/blynktimer.py) and upload these to your device's `/lib` directory (or `/flash/lib`, depending on the firmware. Check `sys.path` on your device).

See https://github.com/thonny/thonny/wiki/MicroPython#managing-other-files for instructions.

Please let me know if you managed to get it working!

aivarannamaa, 2021-04-20:
Please reopen and add more details if you're still stuck!

closed 2021-04-20 by aivarannamaa