thonny/thonny #1728: Thonny - Plugin not visible on Mac

KAGTechnical, 2021-03-24:
Start using the "Get Started with MicroPython on Raspberry Pi Pico" unfortunately when I tried to use the 'machine' and 'uTime' plugins they could not be found as per the error code below. What am I doing wrong?

>>> %Run 'Test 1.py'
Traceback (most recent call last):
  File "/Users/gavinc/Desktop/Test 1.py", line 1, in <module>
    import machine
ModuleNotFoundError: No module named 'machine'

aivarannamaa, 2021-03-24:
It looks like you are trying to run your code on CPython, not on Pico.

Did you change the interpreter from the button on the lower-right corner of Thonny's window?

KAGTechnical, 2021-03-24:
Thank you, that worked.

Reply to: thonny/thonny ***@***.***>
Date: Wednesday, 24 March 2021 at 09:03
To: thonny/thonny ***@***.***>
Cc: KAGTechnical ***@***.***>, Author ***@***.***>

closed 2021-03-24 by aivarannamaa