thonny/thonny #1731: Add "Reconnect" link to Shell when connection gets lost

aivarannamaa, 2021-03-24:
It would be leaner than "Stop / Restart":

* With serial MicroPython it wouldn't soft-reboot
* With SSH it would show the REPL again after process ends but SSH connection is still intact (eg. back-button is pressed on EV3)

aivarannamaa, 2022-04-01:
@bnelsey, you could also try opening the interpreter-switcher (lower-right corner of the main window) and clicking on the menu item, which is already selected.

bnelsey, 2022-04-02:
> @bnelsey, you could also try opening the interpreter-switcher (lower-right corner of the main window) and clicking on the menu item, which is already selected.

woah this works, thanks a lot!
really unintuitive but it definitely does exactly what I needed to do, gotta write this down somewhere