thonny/thonny #1734: [Feature Request] external modification: yes to all

sevku, 2021-03-25:
In the External Modification dialog, it should be possible to say "Yes to all".

aivarannamaa, 2021-03-28:
Thank you for the suggestion!

At the moment I don't see the need for it, but I'd like to understand your viewpoint better. Do you regularly perform external modification on several files so that it becomes tedious to click "Yes" several times? Why?

sevku, 2021-03-29:
> At the moment I don't see the need for it, but I'd like to understand your viewpoint better. Do you regularly perform external modification on several files so that it becomes tedious to click "Yes" several times? Why?

Whenever I pull from git and a lot of files are open, I will have to click through many of those windows.

aivarannamaa, 2021-03-29:
Oh, I see... good point!

aivarannamaa, 2023-07-07:
I'm thinking about adding a new configuration checkbox "Always confirm external modifications" under "Editor" section. If unchecked (by default), the confirmation will be presented only for files which have unsaved changes in Thonny.