thonny/thonny #1738: Thonny 3.2.7 has an issue in my laptop linux. Some directory is missing

JosSom, 2021-03-26:
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1883, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 429, in cmd_stop_restart
    self.restart_backend(True)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 557, in restart_backend
    self._proxy = backend_class(clean)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 1254, in __init__
    super().__init__(clean, get_interpreter_for_subprocess(executable))
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 1062, in __init__
    super().__init__(clean, executable)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 771, in __init__
    self._start_background_process(clean=clean)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 781, in _start_background_process
    env = get_environment_for_python_subprocess(self._executable)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 1393, in get_environment_for_python_subprocess
    overrides = get_environment_overrides_for_python_subprocess(target_executable)
  File "/usr/lib/python3/dist-packages/thonny/running.py", line 1439, in get_environment_overrides_for_python_subprocess
    if os.path.samefile(
  File "/usr/lib/python3.8/genericpath.py", line 100, in samefile
    s1 = os.stat(f1)
FileNotFoundError: [Errno 2] No such file or directory: ''

aivarannamaa, 2021-03-27:
Thank you for the report!

Can it be that you had configured Thonny to use an interpreter which doesn't exist anymore? Please see "Tools => Options => Interpreter"

aivarannamaa, 2021-04-20:
Please reopen if you're still stuck and you have more details to add!

closed 2021-04-20 by aivarannamaa