thonny/thonny #1739: Am I missign something??

toLucky81, 2021-03-26:
Just installed thonny

I just had my new Pico, and I booted it with the boot-button pressed. Downloaded and moved the CircuitPython 6.2.0-beta.4 UF2 file to the Pico. The Pico rebooted and showed me the CIRCUITPY drive with the code.py file. So fare so good.

Now, I wanted to start writing code, so I installed thonny.

went into Tools->Interpreter and switched to CircuitPython (generic) and under port selected the "Pico - CircuitPython CDC control (/dev/ttyACM0)"

wrote the code:
print("Hello Word!")
and saved it as main.py on the Pico

but now thonny keeps printing the msg:
/******************************************/
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 1, in <module>
    from thonny.common import (
ImportError: cannot import name 'InterruptCommand' from 'thonny.common' (/usr/local/lib/python3.8/dist-packages/thonny/common.py)

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
/******************************************/

I tried clicking the "STOP" sign, but it just re-starts and prints the error again.

ran
sudo apt update
sudo apt upgrade
ran perfectly

re-ran thonny - but still the same.

Ubuntu
Description:	Ubuntu 20.10
Release:	20.10
Codename:	groovy

Python installs
whereis python2
python2: /usr/lib/python2.7

whereis python3
python3: /usr/bin/python3.8 /usr/bin/python3.8-config /usr/bin/python3 /usr/lib/python3.8 /usr/lib/python3.9 /usr/lib/python3 /etc/python3.8 /etc/python3 /usr/local/lib/python3.8 /usr/include/python3.8 /usr/share/python3 /usr/share/man/man1/python3.1.gz

Other installs
Visual Studo Code
Putty
Blender
Discord
GIMP
- Nothing that I think have an influince on this install...

Any ideas?

Thanks

Dennis

toLucky81, 2021-03-26:
uninstalled thorny
and reinstalled via bash - ... now it works...

aivarannamaa, 2021-03-27:
Thank you for the report!

From the error it looks as if you had two Thonnys:

* /usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py
* /usr/local/lib/python3.8/dist-packages/thonny/common.py

Could it be that you had pip-installed and apt-installed Thonny?

toLucky81, 2021-03-31:
That may have been the case.

closed 2021-03-31 by toLucky81