thonny/thonny #1741: Error en la ayuda

alito56, 2021-03-27:
Traceback (most recent call last):
  File "/home/alito/apps/thonny/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/tktextext.py", line 229, in dispatch
    return fun(event)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/tktextext.py", line 541, in perform_tab
    return self.perform_midline_tab(event)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/plugins/autocomplete.py", line 321, in patched_perform_midline_tab
    _handle_autocomplete_request_for_text(text)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/plugins/autocomplete.py", line 309, in _handle_autocomplete_request_for_text
    text.autocompleter.handle_autocomplete_request()
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/plugins/autocomplete.py", line 58, in handle_autocomplete_request
    filename=self._get_filename(),
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/running.py", line 268, in send_command
    response = self._proxy.send_command(cmd)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/__init__.py", line 184, in send_command
    return super().send_command(cmd)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/running.py", line 961, in send_command
    self._send_msg(cmd)
  File "/home/alito/apps/thonny/lib/python3.7/site-packages/thonny/running.py", line 964, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
AttributeError: 'NoneType' object has no attribute 'stdin'

aivarannamaa, 2021-03-27:
Thank you for the report!

Do I understand correctly that you were invoking code completion (with TAB key) when you received this error?

Which interpreter were you using this time (regular Python 3 or MicroPython)?

aivarannamaa, 2021-04-20:
Please reopen if you're still stuck and you have more details to add!

closed 2021-04-20 by aivarannamaa