thonny/thonny #1742: raspberry pico version doesn't reload layout or  reopen files from last session

dalemcbeath, 2021-03-28:
using Micropython (Raspberry Pi Pico) interpreter
with single instance, reopen all files, and Raspberry Pi Dark / Tomorrow Night Eighties theme

When I shut down v3.3.6 with open (saved) files and outline pane pulled to the far right and re-open, and re-open, Thonny starts up with the outline pane wide from center to right side, and no files.

Also get strange errors if running PWM PIOs and can't stop the run except with multiple ^D's, but I'll report them separately.

thanks

aivarannamaa, 2021-03-31:
Thank you for the report!

Does it happen all the time? Which OS are you using?

dalemcbeath, 2021-04-02:
new info:  Panes re-open at last used configuration IF plot window is open. If plot is closed, next time program opens, previously resized panes are in default positions.

aivarannamaa, 2021-06-13:
Thank you for the extra details!

I could confirm this both on Windows and Linux. I also found out that it only happens when the window is maximized. In Windows it helps to have another tab on the right-hand pane open, in Linux it doesn't.

It must be a quirk of the UI toolkit. I'll see what I can to get consistent behavior.

aivarannamaa, 2024-05-30:
The layout of the main window has been greatly reworked for the upcoming 5.0 and this problem seems to be fixed there.