thonny/thonny #1743: WARNING:root:Unexpected echo.

dradux, 2021-03-28:
I get the following upon connecting to my device:
```

```
If I hit enter I get the following:
```
WARNING:root:Unexpected echo. Expected b'\r\n', got b'\r\r'

ERROR
```
See screenshots for better representation of error.
[image]

No other logs or info that I can find anywhere. 

The device I'm connecting to is a AITRIP D1 WeMos ESP8266 (ESP-12F). The above is with the MicroPython (ESP8266) Interpreter but I get the same results with any interpreter.

Note: After connecting the ESP8266 via USB I am able to see it with lsusb:

```
Bus 001 Device 012: ID 1a86:7523 QinHeng Electronics CH340 serial converter
```
Fresh install of thonny (3.3.6) via pip in a virtualenv running python 3.8.6.

Any help would be greatly appreciated!

aivarannamaa, 2021-03-28:
Thank you for the report!

Does the device has its own special MicroPython firmware or you're using the one from micropython.org?

What happens when you press Ctrl+C instead of ENTER?

dradux, 2021-03-28:
Hi @aivarannamaa, thanks for the quick response. The issue was I did not flash MicroPython firmware first! After flashing MP to the board with esptool I can successfully connect.

This ticket might be useful to others new to MicroPython - also might be useful to mention this in thonny Assistant or shell if it can be easily/generically caught.

closed 2021-03-28 by dradux