thonny/thonny #1744: Clash with Qgis

pkparchure, 2021-03-30:
The problem is not with Thonny itself. On the contrary I have found Thonny as very good ide as I am learnig Python at present. The problem is like this.

I also have Qgis instilled on my windows 10 system. You may be aware that Qgis depends on Python 37. 
Its started giving trouble after I installed Thonny/ python 37.
Qgis has a plugin called "processing". It started giving error like 

Couldn't load plugin 'processing' due to an error when calling its classFactory() method 
OSError: [WinError 193] %1 is not a valid Win32 application.

Since I had recently installed Thonny and installed several packages, started removing the packages one by one.
Luckily after removing some packages I found that the culprit was the NumPy package. 
As soon as I un-installed NumPy, the Qgis Processing Plugin started working. 
Now that is only temporary solution as I need to use NumPy within Thonny.
Any solution? Kindly let me know

Regards

Pradeep

aivarannamaa, 2021-04-02:
Thank you for the report!

I assume Qgis is 64-bit application and comes with 64-bit Numpy. Thonny is distributed with 32-bit Python and when you installed Numpy, you got 32-bit Numpy.

The problem is that Thonny's package manager installs packages with `--user` switch, ie. they end up in a directory where other interpreters (eg. the one used by Qgis) see it. Furthermore, this location takes precedence over libraries in interpreter-specific directories. So it's likely that Qgis, while trying to import `numpy`, actually tries to import the 32-bit one, which doesn't work (The error `%1 is not a valid Win32 application` is misleading -- it actually means that the library is unsuitable).

The solution is to unistall numpy (and all dependent packages, eg. `pandas` or `matplotlib`) and reinstall them differently.

One option is to install them into Thonny's Python without `--user` switch. For this, go to command line (Tools => Open system shell) and enter `pip install numpy`. The downside of this solution is that when you upgrade Thonny, you need to do this again.

Better solution is to create a virtual environment for your project and install your packages there. Thonny can help you here: for creating the venv, select  "Run => Select intepreter => Alternative Python 3 .... => New virtual environment". When done and the virtual environment is selected (you see its path in the lower-right corner of Thonny's main window), you can use package manager as before, but now the packages go to your virtual environment, where they don't bother Qgis.

pkparchure, 2021-04-03:
Thanks a lot for your prompt reply, I am new to Python and I am learning it now at an age of 62. Thanks to Thonny, I am finding it possible. 
As regards the problem in question. I already found the solution by creating a virtual environment in separate folder. And it is working very well. 
Recently I tried to install VTK, the visualisation tool kit (vtk.org) and the problem is that the VTK is available only as 64bit package for windows.
I think the Thonny could have option for 64 bit Python also..
Next I may try to create another virtual environment using 64 bit python - may be from Qgis.
I will report you the outcome.
Thanks again and regards.
Pradeep

aivarannamaa, 2021-04-04:
I'm glad you found the solution.

Regarding 64-bit only libraries -- one solution is to install 64-bit Python and have Thonny use it as the interpreter for running your programs (ie. Thonny's bundled 32-bit Python will only drive the UI, but it uses another Python for your programs). Use "Run => Select interpreter" or the back-end switcher in the lower-right corner of Thonny's main window.

closed 2021-04-20 by aivarannamaa