thonny/thonny #1745: Thonny startup error

Rdoh97, 2021-03-31:
Thonnystartup error 
Invalid literal for int() with base 10:
please help

aivarannamaa, 2021-03-31:
Thank you for the report!

Did you get this error the first time you started Thonny or have you been able to use Thonny on that computer before? If you have used it successfully before, then can you recall what you changed before it broke?

Rdoh97, 2021-03-31:
I have been able to use thonny before. As I recall the only change I've done was changing the interpreter from rasberry pi pico to default interpreter and tried to have an output of even numbers.

aivarannamaa, 2021-03-31:
Which OS are you using?

If Windows, then please open Windows Explorer, click on its "address bar" (the box which displays current directory) and enter `%LOCALAPPADATA%' there (paste and press ENTER). After this you should see a directory, which is supposed to contain another directory named `thonny-<your username>` (for me it would be `thonny-aivar`).

Please go into this directory and see if it contains file named `ipc.sock`. If it does, then please post it here. (If you want, you can open it with Notepad to see that there is nothing private in it)

aivarannamaa, 2021-03-31:
No, `%appdata%` is another directory. Did you try `%LocalAppData%` (case doesn't matter)?

Rdoh97, 2021-03-31:
I found it how should I proceed to open it with?

Rdoh97, 2021-03-31:
I opened it with note pad and nothing was shown it was simply blank

aivarannamaa, 2021-03-31:
Thank you for the information! 

The file is not supposed to be blank, but I'll investigate.

You should be able to start Thonny after you delete this file.

Rdoh97, 2021-03-31:
thank you and it did work after deleting rpc.sock file

aivarannamaa, 2021-03-31:
Great! Now it's my job to find out how can this file become blank and make sure it won't happen in the next versions.

Rdoh97, 2021-03-31:
Yes please, Yea not sure if this information would help but I did also attempt to open it with enternet explorer and it just kept closing the tab automatically

Would you have any suggestions on what i should do or keep an eye out of to avoid having this problem again?

aivarannamaa, 2021-04-02:
Unfortunately no, at the moment I don't have any ideas what may have caused it.

aivarannamaa, 2022-04-23:
Closing, as I can't reproduce and the work-around (catching the exception and deleting corrupted rpc.sock) is good enough.

closed 2022-04-23 by aivarannamaa