thonny/thonny #1747: Cannot install micropython plug in from package manager or manually

TheTinkerFairy, 2021-04-01:
I have tried and tried and have had no success to install a micropython library to my recent install of Thonny so I can tinker with my Raspberry Pi Pico. Thonny works fine and I can interact with the Pi Pico no problem. But when I try and install a library I always get an error. Looks like the random folder it creates for the install with pip doesn't exist. I tried a re-install of Thonny, running pip direct from the command line but nothing seems to work. I am on a Win 10 PC and installed using "run as administrator", I have tried running Thonny as administrator but that seems to have no effect.

Errors I get are listed below:
ATTEMPT 1 USING THONNY PACKAGE MANAGER
Processing c:\users\user\appdata\local\programs\thonny\lib\site-packages\thonny\plugins\micropython\micropython-ssd1306-master.zip
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\user\AppData\Local\Programs\Thonny\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-req-build-jdosp_wt\\setup.py'"'"'; __file__='"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-req-build-jdosp_wt\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\user\AppData\Local\Temp\pip-pip-egg-info-6tbw7hzz'
         cwd: C:\Users\user\AppData\Local\Temp\pip-req-build-jdosp_wt\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\user\AppData\Local\Temp\pip-req-build-jdosp_wt\setup.py", line 7, in <module>
        import sdist_upip
    ModuleNotFoundError: No module named 'sdist_upip'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Process returned with code 1

ATTEMPT 2 USING THONNY PACKAGE MANAGER AFTER RE-INSTALL
C:\Users\user\AppData\Local\Programs\Thonny\python.exe -m pip install --no-cache-dir --user micropython-ssd1306 --disable-pip-version-check
Collecting micropython-ssd1306
  Downloading micropython-ssd1306-0.3.tar.gz (1.8 kB)
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\user\AppData\Local\Programs\Thonny\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-_dmpdcgv\\micropython-ssd1306\\setup.py'"'"'; __file__='"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-_dmpdcgv\\micropython-ssd1306\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\user\AppData\Local\Temp\pip-pip-egg-info-gb5k6oj1'
         cwd: C:\Users\user\AppData\Local\Temp\pip-install-_dmpdcgv\micropython-ssd1306\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\user\AppData\Local\Programs\Thonny\lib\tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-_dmpdcgv\\micropython-ssd1306\\setup.py'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error: process returned with code 1

ATTEMPT 3 RUNNING MANUALLY FROM SYSTEM SHELL
C:\Users>pip install micropython-ssd1306
Collecting micropython-ssd1306
  Downloading micropython-ssd1306-0.3.tar.gz (1.8 kB)
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\user\AppData\Local\Programs\Thonny\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-2grvv0tx\\micropython-ssd1306\\setup.py'"'"'; __file__='"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-2grvv0tx\\micropython-ssd1306\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\user\AppData\Local\Temp\pip-pip-egg-info-e5uj5g5i'
         cwd: C:\Users\user\AppData\Local\Temp\pip-install-2grvv0tx\micropython-ssd1306\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\user\AppData\Local\Programs\Thonny\lib\tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-2grvv0tx\\micropython-ssd1306\\setup.py'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 20.1.1; however, version 21.0.1 is available.
You should consider upgrading via the 'C:\Users\user\AppData\Local\Programs\Thonny\python.exe -m pip install --upgrade pip' command.

ATTEMPT 4 USING COMMAND INSIDE THONNY (TO BE HONEST I WAS JUST RANDOMLY  TRYING STUFF AT THIS POINT!)
C:\Users\user\AppData\Local\Programs\Thonny\python.exe -m pip install --no-cache-dir --user micropython-ssd1306 --disable-pip-version-check
Collecting micropython-ssd1306
  Downloading micropython-ssd1306-0.3.tar.gz (1.8 kB)
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\user\AppData\Local\Programs\Thonny\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-e7tguv44\\micropython-ssd1306\\setup.py'"'"'; __file__='"'"'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-e7tguv44\\micropython-ssd1306\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\user\AppData\Local\Temp\pip-pip-egg-info-zs953e69'
         cwd: C:\Users\user\AppData\Local\Temp\pip-install-e7tguv44\micropython-ssd1306\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\user\AppData\Local\Programs\Thonny\lib\tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\user\\AppData\\Local\\Temp\\pip-install-e7tguv44\\micropython-ssd1306\\setup.py'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error: process returned with code 1

Please can someone advise if this is a bug or my own issue (or dumb moment).

aivarannamaa, 2021-04-02:
Thank you for the report!

Do I understand correctly, that at 1st attempt you tried to install it from zip file? If you just downloaded the code from the GitHub, then this wouldn't do -- those zip-s that are on PyPI have a bit different layout.

Second and fourth attempts seems to use regular `pip`, ie you probably didn't have MicroPython back-end selected at this time or you used "Tools => Manage plug-ins" instead of "Tools => Manage packages".

With attempt 3 you explicitly used regular `pip`, but this doesn't work for micropython packages.

So please make another attempt:

* Make sure you have MicroPython interpreter selected
* Select "Tools => Manage packages"
* Paste `micropython-ssd1306` into search box and press ENTER
* Click on the first match
* Click install

In my Windows 10 machine it failed first time, but I suspect it was a glitch in communicating with Pico. When I tried again, I could install this package.

I'll try to make Thonny warn when somebody makes these mistakes again.

TheTinkerFairy, 2021-04-02:
Thanks so much for the quick response. It looks like it was my misunderstanding in installing a package and not a plug-in. Oddly the micropython-ssd1306 appears as a plug-in in the plug-in manager nothing is listed in my installation in the left pane which added to my confusion. All working now and the screen works now the library is installed.
Could Thonny detect the missing import library and do a guided or wizard install?

aivarannamaa, 2021-04-02:
> Could Thonny detect the missing import library and do a guided or wizard install?

It's not very easy to do, as module names don't need to match with PyPI names. Also, two different PyPI distributions may contain module with same name; sometimes it is important to install specific version of the package and so on. MicroPython packaging is still very messy, so it would complicate matters even more.

closed 2021-04-02 by TheTinkerFairy