thonny/thonny #1748: One Dark theme port for Thonny

ranelpadon, 2021-04-02:
**One Dark** is a very popular theme introduced by Atom, then ported to various apps. VS Code is the most popular text editor/IDE now in web development. And the **Top 1** and **Top 3** color schemes in VS Code are ports of One Dark, with a combined **5.5 million downloads**!

[image]

https://marketplace.visualstudio.com/search?target=VSCode&category=Themes&sortBy=Installs

Personally, I love One Dark very much, and use it as my color scheme in all my programming-related apps: `Alacritty`, `Tmux`, `Vim`, `VS Code`, `PyCharm`, `Sublime Text`, `iTerm`, and `Kitty`. I work as a full-time software engineer, and part-time lecturer in a university. This year, I started using the **Thonny** IDE in our introductory programming class, my students like it as well since it's lightweight, has clean UI, has powerful program logic/variable inspector, etc. Although I kinda miss the One Dark color when using it. Eventually, I glimpsed about the **Dracula** theme in Thonny home page, and got the idea of implementing similar one for One Dark. So, here's the result:

[image]
 
It's already indexed in **PyPI** as `thonny-onedark`:
https://pypi.org/project/thonny-onedark/

and works as intended in my local when installing from PyPI:

[image]

GitHub repo:
https://github.com/ranelpadon/thonny-onedark

Special thanks also to @danspinola, got the idea of the implementation from https://github.com/thonny/thonny/issues/1304. Similarly, I filed this issue just to let you know, and hopefully will be included also in the **Thonny homepage** since One Dark is really popular nowadays. :)

aivarannamaa, 2021-04-02:
That's great! I added the link to Thonny's homepage and mentioned it in Twitter: https://twitter.com/thonnyide/status/1378058229809348610

ranelpadon, 2021-04-02:
@aivarannamaa Yay, I've seen it, you could also mention my handle (**`@_ranelpadon`**) in your Twitter post, since I've lost my `@ranelpadon` account years ago. Cheers :)

ranelpadon, 2021-04-02:
Just realized that the docstrings' indention is not good, so I updated the screenshot:
[image]

I posted it also in Python's Reddit:
https://www.reddit.com/r/Python/comments/mist7k/one_dark_theme_for_thonny_ide/

closed 2021-04-06 by ranelpadon