thonny/thonny #1750: Device is busy or does not respond

Keu99, 2021-04-02:
[image]

aivarannamaa, 2021-04-02:
Thank you for the report!

Did you try the suggestions given in the message?

Do you have anything in your main.py?

Keu99, 2021-04-03:
I try it even unistalling and install thony again, but only when I tried the interpreter for Raspberry Pi Pico, Thony can't detect the port for Raspberry

aivarannamaa, 2021-04-03:
According to your screenshot, Thonny has connected to Pico, it's just that Pico doesn't answer. This can be because a script is running on your Pico.

Have you used MicroPython with this Pico before? If no, then have you installed MicroPython for it?

Keu99, 2021-04-03:
I have not used MicroPython with this Pico, I mean even when Pico is disconnected Thony throw the same error.

aivarannamaa, 2021-04-04:
To be sure, are you trying to use Thonny for MicroPython on Pico or for regular Python 3?

If you want't to use MicroPython, then you need to first install MicroPython on your Pico.

> even when Pico is disconnected Thony throw the same error.

When you disconnect Pico, the error is not supposed to go away. What were you trying to do with Thonny after you disconnected Pico?

aivarannamaa, 2021-04-20:
Please reopen if you're still stuck and you have more details to add!

closed 2021-04-20 by aivarannamaa

zrl4836, 2021-12-20:
下载 闪存重置 文件 重新 烧录进去 即可解决 https://datasheets.raspberrypi.com/soft/flash_nuke.uf2