thonny/thonny #1753: Problem with the USB port

RafaRivera, 2021-04-06:
I'm installing Raspberry Pi Pico through a PC with Windows 7. I connected and I can see that is recognized  as store device RP1-RP2.
The problem  appear when a trying of install de OS. Windows cut off and Thonny send a messages that it does'n find the serial port.

It only appear two options as COM ports 1 and 2 and when I connect an Arduino board the system assign COM7. 

What can I do?

aivarannamaa, 2021-04-09:
Thank you for the report!

Can you copy the whole error message for me? Did you select specific port or you let Thonny to find it automatically?

BTW, I've never managed to properly connect to Pico on Windows 7.

aivarannamaa, 2021-04-20:
Please reopen if you're still stuck and you have more details to add!

closed 2021-04-20 by aivarannamaa