thonny/thonny #1754: thonny的micropython文件传输协议

Spring641, 2021-04-07:
请问有thonny和micropython的文件传输协议文档吗？我使用pyb文件传输的时候出错。不知道他们直接通讯规则是怎么样的

Spring641, 2021-04-07:
[image]

aivarannamaa, 2021-04-09:
I see you had trouble opening a file from the device. Does it happen all the time?

xfbluesky, 2021-04-11:
I've been having this problem since today, about 80 percent probability

Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/base_file_browser.py", line 563, in on_double_click
    self.open_file(path)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/base_file_browser.py", line 1008, in open_file
    get_workbench().get_editor_notebook().show_remote_file(path)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/editors.py", line 949, in show_remote_file
    return self.show_file(make_remote_path(target_filename))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/editors.py", line 923, in show_file
    editor = self.get_editor(filename, True)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/editors.py", line 1012, in get_editor
    return self._open_file(filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/editors.py", line 996, in _open_file
    if editor._load_file(filename):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/editors.py", line 169, in _load_file
    result = self._load_remote_file(filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/editors.py", line 228, in _load_remote_file
    raise RuntimeError(response["error"])
RuntimeError: Problem with a management command

SCRIPT:
__thonny_helper.print_mgmt_value(__thonny_fp.read(1024))

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 21, in print_mgmt_value
MemoryError: memory allocation failed, allocating 1768 bytes

Spring641, 2021-04-12:
> 我看到您无法从设备打开文件。它会一直发生吗？
是的，不过我最后查找是我的设备收发数据出问题了

Spring641, 2021-04-12:
还有一个问题，向pyb设备发生文件(1MB)左右的文件的时候，在中图取消发送文件，不过pyb还是一直再写入文件。貌似是点击取消按钮没有完全关闭文件流。文件在后台还在传输

aivarannamaa, 2021-04-15:
I'm getting quite confused now. Please post each issue separately and please write in English, Estonian, Spanish or French.

Regarding the MemoryError -- can you tell me how large is the file you are trying to open? Please also try following:

* Close all files
* Click on the red Stop/Restart button
* Now try opening the same file

Does it work now?

aivarannamaa, 2021-06-13:
Please reopen if you have more details to add.

closed 2021-06-13 by aivarannamaa