thonny/thonny #1755: [Feature request] Print info if the device is busy

bitboy85, 2021-04-07:
If the device is already executing code thonny does not show its files or some output in the commandline.
The os filemanager still allows editing the files on the device.

I think it would be helpful to print a line like "The device is currently running. Press CTRL + C to abort" or something like this in the comandline line.

aivarannamaa, 2021-04-08:
Thank you for the suggestion!

Did you get anything in the output when your device was running and you connected Thonny to it? Actually Thonny is supposed to print this information, but only if the device doesn't print anything in 2.5 seconds: https://github.com/thonny/thonny/blob/707e69ec3a567df5f82205c5a2ae0d79f186ed25/thonny/plugins/micropython/bare_metal_backend.py#L820

bitboy85, 2021-04-08:
The Output is just an empty line.
Hitting Enter will give you an error message "WARNING:root:Unexpected echo. Expected b'\r\n', got b''"
[image]

aivarannamaa, 2021-04-09:
Ok, I should then consider only visible characters when deciding whether to print the advice or not.

aivarannamaa, 2021-11-23:
Thonny 4.0 will print the "device is busy or doesn't respond" message when no visible chars appear after certain time after connection or when interrupt doesn't have an effect.

closed 2021-11-23 by aivarannamaa