thonny/thonny #1756: STOP button ambiguity

IhorNehrutsa, 2021-04-08:
The issue is that the 'STOP' button (and the 'Stop/Restart backend Ctrl+F2' command)  
doesn't stop the execution of a program on the device, it only restarts the backend(probably).  
This code illustrates the issue:  
[test_finally.py.txt](https://github.com/thonny/thonny/files/6279301/test_finally.py.txt)  
```
# test_finally.py
from time import sleep
from sys import exit

try:
    print("try: Start hardware, open sockets etc!")
    i = 0
    while True:
        i += 1
        if i > 10:
            exit(0)
        print(i, end=' ')
        sleep(1)
except KeyboardInterrupt as e:
    print("")
    print("except:", type(e), ':', e)
except SystemExit as e:
    print("")
    print("except:", type(e), ':', e)
finally:
    print("finally: Stop hardware, close sockets etc!")
```  
[image]  
The 'finally:' section doesn't execute if the 'STOP' button is pressed.  
The hardware like Timers, PWM, RMT still works.  
This dissonances with the button name.  

Also, it works the same on MicroPython(ESP32).  
Thanks for your work.

aivarannamaa, 2021-04-09:
Thank you for the report!

Here's what the STOP button does with MicroPython back-end:

* kill the Python subprocess which is talking to the device over serial (ie. this includes closing the serial connection)
* start new subprocess, which is creating new serial connection
* Send Ctrl+C and Ctrl+A to the device (ie. interupt the work and enter RAW repl) **and consume all output** -- that's why you didn't see `finally: Stop hardware, close sockets etc!`. Try writing a file in the finally block and you'll see that the block is executed just like when you did Ctrl+C manually. The idea was to provide means for getting a "clean sheet". I might reconsider this choice, as showing the effect of interrupt would make the process more transparent.
* Finally Thonny sends Ctrl+D to the device to initiate soft reboot (ie. restarting MicroPython VM) without running main.py (main.py is ignored when soft rebooting at raw repl)) in order to get clean environment.

About stopping hardware -- this is device specific and I'm not sure I want to worry about this. What I could do is offering a choice for doing a `machine.reset()` instead of soft reboot, but after this the device would run main.py.

Have you tried stopping the hardware in your finally block?

IhorNehrutsa, 2021-04-09:
I changed the test_finally.py to examine the hardware in the finally block.  
[test_finally.py.txt](https://github.com/thonny/thonny/files/6286538/test_finally.py.txt)   
```
from time import sleep
from sys import exit, print_exception
from machine import Pin, PWM, Timer

try:
    print("try: Start hardware, open sockets etc!")
    LED = 2  # (GPIO02)
    led = Pin(LED, Pin.OUT, value=1)
    print("PWM 1Hz")
    pwm = PWM(led, freq=1, duty=512)  # freq=1Hz  # duty=512 == 50%

    i = 0
    while True:
        i += 1
        if i > 10:
            exit(0)
        print(i, end=' ')
        sleep(1)
except KeyboardInterrupt as e:
    print("")
    print("except:", type(e), ':', e)
except SystemExit as e:
    print("")
    print("except:", type(e), ':', e)
finally:
    print("finally: Stop hardware, close sockets etc!")
    pwm.deinit()
    
    print("PWM 10Hz")
    pwm = PWM(led, freq=10, duty=512)  # freq=10Hz  # duty=512 == 50%
    sleep(2)  #  Unexpected KeyboardInterrupt raises here! PWM 10Hz still works!
    pwm.deinit()
    
    print("led on")
    led.value(1)
    sleep(2)
    print("led off")
    led.value(0)
    
    print("finally: Done!")
```
PWM LED blinks 1Hz in the main loop,  
PWM LED blinks 10Hz in the finally block and then  
the LED will light up for 2 seconds and turn off.  

When Ctrl-C is pressed shortly ones(!) in main loop the KeyboardInterrupt blok works as expected, 
then the finally block really starts, but it is breaked by an unexpected secondary(repeated) KeyboardInterrupt.  
The finally block does not execute completely.  
It seems Thonny sends additional Ctrl-C to the device and this related to the #1757.  
[image]  

Thanks for your work.

P.S. I tested test_finally.py in
https://github.com/jungervin/EsPy 
and there is not secondary KeyboardInterrupt.

aivarannamaa, 2021-04-09:
You're right, Thonny sends two Ctrl-C-s to be sure. It never occurred to me to wait a bit to let the device pack its things. I'm going to deal with this under #1757

closed 2021-06-13 by aivarannamaa