thonny/thonny #1757: Don't issue two Ctrl+C-s too quickly in row in order to give time for finally-blocks to clean up stuff

aivarannamaa, 2021-04-09:
See #1756

LDmicroGitHub, 2021-04-10:
Some idea: You may send to REPL not clear user code but covered or closured by your own code.
Your code may send a spatial message back to front from REPL.
```
##########################
# Beginning of the code which sent to REPL
# Thonny's code prior user code
do something  
    ##########################
    # user code begin 
    try:
        while True:
            print(1, end=' ')  # first Ctrl-C will interrupt normal user loop
        while True:
            print(2, end=' ')  # never will work
    finally:    
        while True:
            print(3, end=' ')  # second Ctrl-C will interrupt the user's final mad
        while True:
            print(4, end=' ')  # never will work
    # user code end
    ##########################
# Thonny's code after user code    
do something 
print user_finally_block_is_completed_mesasage to the Thonny's front from the REPL
# End of the code which sent to REPL
##########################
```

aivarannamaa, 2021-04-15:
@LDmicroGitHub, I'm afraid I didn't understand your suggestion. What do you propose Thonny should do in the last block?

IhorNehrutsa, 2021-04-15:
It seems that the second Ctrl-C is needed only in this case:
This code requires two Ctrl-C
```
##This code requires two Ctrl-C:
try:
    while True:
        pass
finally:    
    while True:
        pass
```
But this is a strange(not typical) code, just like the code below.
This code requires three Ctrl-C
```
##This code requires three Ctrl-C:
try:
    while True:
        pass
finally:    
    try:
        while True:
            pass
    finally:    
        try:
            while True:
                pass
        finally:    
            pass
```

The code may be like this:
```
##########################
# Beginning of the code which sent to REPL
# Thonny's code prior user code
#do something
if True: # this line is only for make indents
    ##########################
    # user code begin 
    try:
        from time import sleep
        print("main: user init hardware")
        while True:
            #main user loop
            print(1, end=' ')  # first Ctrl-C will interrupt main user loop
            sleep(1)
    except KeyboardInterrupt as e:
        print("")
        print("except:", type(e), ':', e)
    finally:    
        print("finally: user deinit hardware")
        for _ in range(3):
            print(2, end=' ')  # some time to work
            sleep(1)  # some time to work
        print("")
        print("finally: done!")
    # user code end
    ##########################
# Thonny's code after user code    
print("user_finally_block_is_completed_mesasage to the Thonny's front from the REPL via the serial connection")
#do something 
# End of the code which sent to REPL
##########################

```
I am not sure about understanding the REPL (raw mode vs copy-paste mode) and Thonny(back vs front). May you give me some links?
If I truly understand the REPL is a black box for Thonny. Thonny doesn't know about the execution of the user code.
Thonny can add 
print("user_finally_block_is_completed_mesasage to the Thonny's front from the REPL via the serial connection")
after user code. 

The main idea is that if 
Thonny catch the 'user_finally_block_is_completed_mesasage' (suppress it output for user)
then don't need to send a second Ctrl-C

I've tested it in REPL paste mode, but I hope it works in raw mode too.  
[image]

aivarannamaa, 2021-11-17:
This part is now refactored and should work correctly. Will be available in the first beta of Thonny 4.0

closed 2021-11-17 by aivarannamaa