thonny/thonny #1761: "Backend terminated or disconnected. Use 'Stop/Restart' to restart. " when use camera with cv

binghe2402, 2021-04-14:
I have the same problem. I run cv2 (4.5.1.48) on my Thonny 3.3.6 on macOS 11.2.3 (20D91) , and when I try to use the camera the error occurred(Backend terminated or disconnected. Use 'Stop/Restart' to restart.)
My code is
```python
import cv2
cap = cv2.VideoCapture(0)
```
if I do not use camera, it works.
```python
import cv2
cap = cv2.VideoCapture("test_video.mp4")
while(True):
    ret , frame = cap.read()
    cv2.imshow('frame',frame)
    if cv2.waitKey(1) &0xFF ==ord('q'):  
    	break
cap.release()
cv2.destroyAllWindows()
```

aivarannamaa, 2021-04-15:
Thank you for the report!

Can you run your code with "Run => Run current script in terminal" ? Do you get any error messages when you run it like this?

binghe2402, 2021-04-16:
Thank you.
I get no error when I run it in terminal.

aivarannamaa, 2023-07-07:
I was able to reproduce this in macOS Ventura with an error message: 
```
OpenCV: not authorized to capture video (status 0), requesting...

Process ended with exit code -9.
```

It looks like the library tried to initiate the permission dialog, but for some reason it did not succeed. When I run the minimal script via Terminal, the permission dialog does pop up but it asks the permission for Terminal, so the grant won't affect Thonny.

At the moment I don't know how to solve this.

chrisbuerginrogers, 2023-11-08:
I believe that you have to add a magic line to the Info.plist so that it triggers the OS to add Thonny to 4th list of apps under camera under security.  I tried and it failed - but then again I am not well versed in the plist world.

aivarannamaa, 2023-11-18:
@chrisbuerginrogers, thanks for the idea! Can you share more details about your attempt?

chrisbuerginrogers, 2023-11-19:
Here is what I want:
Run this code in Thonny (with OpenCV installed - that part works great).

import cv2

vid = cv2.VideoCapture(0)

while(True):
    ret, frame = vid.read()
    cv2.imshow('frame', frame)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

vid.release()
cv2.destroyAllWindows()

When I run that now I get

OpenCV: not authorized to capture video (status 0), requesting...
OpenCV: camera failed to properly initialize!
Traceback (most recent call last):
  File "/Users/crogers/Documents/Work/Classes/2023 2Fall/ME35/Codes/OpenCV_example.py", line 16, in <module>
    cv2.imshow('frame', frame)
cv2.error: OpenCV(4.8.1) /Users/xperience/GHA-OpenCV-Python/_work/opencv-python/opencv-python/opencv/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function ‘imshow'

What I hoped would happen
Error - need to go to Settings:Security:Camera and allow access to Thonny

Here is how I think it is possible (but does not work for me)
Here<https://developer.apple.com/documentation/bundleresources/information_property_list/protected_resources/requesting_authorization_for_media_capture_on_macos?language=objc#> is the best description I could find -  I have tried adding this code below (VSCode was kind enough to supply it).  Clearly I am doing something wrong still.

<key>NSCameraUsageDescription</key>
<string>An application in Visual Studio Code wants to use the Camera.</string>
<key>NSHighResolutionCapable</key>
<true/>

It would be great if we could get camera and mic access - so that it would be easy to do pyaudio etc (My students find VSCode overwhelming).  Thanks a lot.
____________________________________
ph: 617 627 2882  web<https://engineering.tufts.edu/me/people/faculty/chris-rogers>, cal<https://fantastical.app/crogers-bpB2/meetings-w-chris>, and zoom<https://tufts.zoom.us/my/crogers>

Caution: This message originated from outside of the Tufts University organization. Please exercise caution when clicking links or opening attachments. When in doubt, email the TTS Service Desk at ***@***.******@***.***> or call them directly at 617-627-3376.

mithos91, 2024-06-12:
No updates on this right?