thonny/thonny #1762: Python scripts will not run using WebREPL

clrdls, 2021-04-14:
I was using Micropython v1.13 on an ESP8266 with Thonny 3.3.3 on a Raspberry Pi4 and WebREPL was working fine.  I had to move up to an ESP32 VROOM with Micropython v1.14 and Thonny 3.3.3 would not work with it.  I read that a newer version of Thonny would work so I did a software update on the Raspberry Pi and got Thonny 3.3.6.  It will connect with the ESP32 using USB or WebREPL.  With the USB connection everything works fine but with WebREPL python commands in the lower window/shell will execute but if I try to run even a single line micropython program like a print command, I get %Run -c EDITOR_CONTENT like normal but nothing happens.  Nothing prints and the >>> prompt does not return.  If I click on stop, the prompt will return after about 10 seconds.  I also tried using Windows 10 Thonny 3.3.5 and 3.3.6 with the same results.  The ESP8266 also now no longer runs a python program in WebREPL using Thonny 3.3.6.

aivarannamaa, 2021-04-14:
Thank you for the report!

Have you tried waiting a minute after you run your editor content?

Your problem may be similar to #1565. By the request of some users, recent Thonny versions soft-reboot MicroPython VM before running the code from the editor. This means boot.py gets executed and if you are connecting to your WiFi network in it, things become slow. I'll make this optional in the next version #1716.

aivarannamaa, 2021-04-14:
Until the next release you could try the work-around described under #1565

clrdls, 2021-04-15:
Thanks for your reply.  I waited 10 minutes for something to happen but nothing did.  When I click the stop button, it takes 8 to 10 seconds for the cursor to return.  I tried ctrl-d and got WebREPL connected, ctrl-F2 and the file list on the esp8266 went away, ctrl-c and the file list came back along with the cursor.  Tried running the script and got %Run ...  no action and had to click stop to get the cursor back.  Again this was working fine with version 3.3.3 on the ESP8266 but is not working on version 3.3.6 using WebREPL but it works fine through the USB.  I don't believe it has anything to do with restarting or slow action.  boot.py is very fast.

Thanks

aivarannamaa, 2021-04-15:
@clrdls, could you send me your scripts (without secrets, of course) so that I can try to replicate your setup?

clrdls, 2021-04-19:
Aivar

Did you get my reply to this email?  I sent it as a reply and not a 
reply all.

Also, I saw on  line where someone was having a similar problem with 
micropython scripts not running with WebREPL and found that there was a 
conflict with a 10 millisecond timer in his program so he changed his 
timer to 11 milliseconds and it started working.  I don't have any 
timers but there may be one in Thonny.  I do not know what he was using 
to run the program but he did not mention Thonny.

Let me know if you got the reply.

Thanks

aivarannamaa, 2021-04-19:
@clrdls, no I did not receive an e-mail from you after my last comment.

clrdls, 2021-04-19:
Replied to all this time.

These are the scripts on an ESP32 vroom.  Connected to a Raspberry Pi4 
running Thonny 3.3.6,  The ESP32 firmware is v1.14 for a v4 development 
board.

Thanks

aivarannamaa, 2021-04-19:
Sorry, but still no success. I see your reply now, but attachments doesn't work in GitHub when you reply from e-mail. You need to post in GitHub.

clrdls, 2021-04-24:
Aivar

It has been a few days since I posted the scripts you requested on 
github.  Did you get them?  Did you see the same problem that I am 
seeing?  I tried running the scripts on WebREPL.html and they run OK 
there but Thonny is much better for editing and testing scripts.

Thanks

aivarannamaa, 2021-04-24:
Sorry for not replying earlier!

I was able to reproduce with my WROOM, but I don't have a solution yet. 

A work-around is to comment out this line: https://github.com/thonny/thonny/blob/v3.3.6/thonny/plugins/micropython/bare_metal_backend.py#L954 but then you need to take into account that the script is run with the environment (ie. global variables) left by previous run.

You can find Thonny's modules by selecting "Tools => Open Thonny program folder".

aivarannamaa, 2021-04-25:
I believe I now found the reason. The fix wil be in the next release.

closed 2021-04-25 by aivarannamaa

aivarannamaa, 2021-04-30:
Please try the new release: https://github.com/thonny/thonny/releases/tag/v3.3.7