thonny/thonny #1773: cant update packages on thonny 3.3.6 on raspberry pi 4

krischinx, 2021-04-18:
When I try and update packages using the manage packages too, it will show the current package , and the latest package information, but on the bottom, the dialog box for update does not show up.

If the package isn't installed, it will show the install dialog, and this works.

I've tried using tk files instead of zenity in options.
I've tried using different interpreters./ bin/python3, /bin/python3.7, /usr/bin/python3, /usr/bin/python3.7.
I've tried different themes.
I've uninstalled, and reinstalled thonny.

aivarannamaa, 2021-04-19:
Thank you for the report!

Do I understand properly, that you don't see the update buttons? This is expected for packages, which are installed on system level (as opposed to packages installed with `--users` switch, as Thonny does). The reason is that in some cases an application may depend on specific version of a package, so Thonny doesn't want to make it easy to mess up something. Besides, some Python packages on Linux are `apt`-installed, not `pip`-installed.

Which package do you want to update and where is it installed to (Thonny shows this among other package info)? If it's apt-installed and you have up to date system, then the best solution is to create a virtual environment and install a newer version of the package there (Run => Select interpreter => Alternative Python 3 ... => Create virtual environment).

aivarannamaa, 2021-06-13:
Please reopen if you still have questions or have details to add!

closed 2021-06-13 by aivarannamaa