thonny/thonny #1775: problems after update

leddabutz, 2021-04-20:
Hi, 

learning micropython by "The official Raspberry Pi Pico Guide"
i coded the "reaction game" and got the following when i startet the program:
nevertheless the program works sometimes ... maybe a bouncing push button?
sorry, my english is not so fine

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1708, in event_generate
    handler(event)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 433, in _handle_program_output
    self._update_visible_io(None)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 512, in _update_visible_io
    self._apply_io_event(data, stream_name)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 555, in _apply_io_event
    self.active_extra_tags.pop()
IndexError: pop from empty list

aivarannamaa, 2021-04-20:
Thank you for the report!

Which Thonny version are you using and on which platform?

leddabutz, 2021-04-20:
after update i use version 3.3.6 
first i wrote the program like in book, with this version i got the error message you see in my first posting
[reaction_game_fehlerbericht.pdf](https://github.com/thonny/thonny/files/6342054/reaction_game_fehlerbericht.pdf)

leddabutz, 2021-04-20:
my Pico is connected to a raspberry pi 4, 8gb, with the raspberry os (32-bit), ever updated to latest version. so 2 or 3 days ago there was an update for the python-software

aivarannamaa, 2021-04-20:
Thank you for the clarification!

How did you run your code?

* With F5 (or the toolbar button with white triangle)?
* By saving it as main.py and soft rebooting (Ctrl-D)?
* By copy-pasting the code to Thonny's Shell?

aivarannamaa, 2021-04-20:
One more thing - did you see any other errors in the shell before the stacktrace that you gave me? 

What happend, when you close the error dialog? Does your program work properly after this?

leddabutz, 2021-04-20:
Am Tue, 20 Apr 2021 02:36:26 -0700
schrieb Aivar Annamaa ***@***.***>:

> One more thing - did you see any other errors in the shell before the
> stacktrace that you gave me? 
> 
> What happens, when you close the error dialog? Does your program work
> properly after this?
> 

Hi and thank you for your fast reply,

i copied and pasted two textmessages for you, which was shown to me.

The first one in a separate popup-window: "problem when handling
'ProgramOutput'"

The second one in the commandline-window:

You'll find it in attached pdf-file

Three astonishing facts i found out:

- softwaremanager shows me min. 4 versions of Python installed:
	2-2.7.16-1
	2.7-2.7.16-2+deb10u1
	3-3.7.3-1
	3.7-3.7.3-2+deb10u3

- i did run the code by clicking the toolbar button (green, white
  triangle), automaticly the code is saved first

- other error? Not before, but in the following runs.
  in the commandline appears text in blue and red and what miracle,
  between the errortext, fragmented output with correct 'user_name' and
  time in Milliseconds...
  Seems to be the correct output as it was expected

when i closed the error dialog, i clicked the stop button in
thonny (red, white square) and the commandline shows with 3 '>>>'
seemed to be clear. The next try presents another errormessage...

aivarannamaa, 2021-04-23:
Did you find a solution? From my viewpoint the issue is not solved as Thonny is not supposed to give this error message.

leddabutz, 2021-04-23:
At first: There is running Micropython on the RP-pico, so it doesn't matter which version of Python is installed on my Raspi, i think.
I'll check it out. i'm going to deinstall all Python and then reinstall the latest version...
Second: In this book from "Raspberry Pi Press" the student has the job at the end of each chapter to develop the code in a specified direction. So, when i coded a new solution, i'm not so sure how to solve the problem. Maybe my code is not so clean, but i cannot understand the errormessages. When i completely delete my additional part of code there are no more errormessages 
A suspicion is: i used german language for printed output. Like in many other languages we have some special characters on our keyboards. By using them, there may occure interpreterwarnings, cause this isn't a mistake but the software can't handle this?
What do you think?

aivarannamaa, 2021-04-23:
@leddabutz, no, I don't find it likely that non-ascii characters are causing problems, but it would be easy to check -- just remove non-ascii characters from your strings and see if it makes a difference.

You mentioned two error messages, but your pdf has same message as the first comment in this thread. Were they the same? In particular, did you see something that is shown in the first comment under #1586?

leddabutz, 2021-04-23:
in some kind does is look similar, especially the fragmented output, which was infected, with other characters from the error-message. It happenede, when i initiated an output with 

print("Glückwunsch ", + user_name) and 
print("Deine Reaktionszeut war ", + str(timer_reaction), + " Millisekunden")

Now i wrote a part of the first program once more, look what happened:

**This is my code:**

import machine
import utime
import urandom

pressed = False
# user_name = input ("Wie heißt Du? ")
led = machine.Pin(15, machine.Pin.OUT)
button = machine.Pin(14, machine.Pin.IN, machine.Pin.PULL_DOWN)

def button_handler(pin):
    global pressed
    if not pressed:
        pressed=True
        print(pin)
        
led.value(1)
utime.sleep(urandom.uniform(5, 10))
led.value(0)

button.irq(trigger=machine.Pin.IRQ_RISING, handler=button_handler)
____________________________________________________________________

You see the line with the hashtag? In this form the program works proper. On the other hand when i take away the hashtag and user_name is asked, then i do my input like Tom and ENTER, i got the following error...

____________________________________________________________________

Problem when handling ‚ProgramOutput‘

_Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1708, in event_generate
    handler(event)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 433, in _handle_program_output
    self._update_visible_io(None)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 512, in _update_visible_io
    self._apply_io_event(data, stream_name)
  File "/usr/lib/python3/dist-packages/thonny/shell.py", line 555, in _apply_io_event
    self.active_extra_tags.pop()
IndexError: pop from empty list_
_________________________________________________________________________________

Wie heißt Du? Tom
THONNY FAILED TO EXECUTE COMMAND Run

SCRIPT:

class __thonny_helper:
    try:
        import uos as os
    except ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = dict()
    last_repl_values = []
    @classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.last_repl_values.append(obj)
            cls.last_repl_values = cls.last_repl_values[-5:]
            print('[object_link_for_thonny=%d]' % id(obj), repr(obj), '', sep='')

    @staticmethod
    def print_mgmt_value(obj):
        print('<thonny>', repr(obj), '</thonny>', sep='', end='')

    @staticmethod
    def repr(obj):
        try:
            s = repr(obj)
            if len(s) > 50:
                s = s[:50] + "..."
            return s
        except Exception as e:
            return "<could not serialize: " + str(e) + ">"

    @classmethod
    def listdir(cls, x):
        if hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @classmethod
    def getcwd(cls):
        if hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:
Pin(14, mode=IN, pull=PULL_DOWN)

STDERR:

ManagementError
>>>

leddabutz, 2021-04-23:
sorry i don't know that the hashtag is a formatting sign. So the now extrem thick written line have had a leading hashtag

aivarannamaa, 2021-04-25:
Thank you for this information! 

I'll get my Pico with soldered headers soon, so I'll be able to recreate your experiment.

leddabutz, 2021-04-26:
Meanwhile i think the errors causes in different reasons. Look what happens when my input has an uknown character:

>>> %Run -c $EDITOR_CONTENT
Wie heißt Du? Düsentrieb
Dsentrieb
WARNING:root:Unexpected echo. Expected b'D\xc3\xbcsentrieb\r\n', got b''
>>>

aivarannamaa, 2021-04-27:
Thank you for these extra details!

aivarannamaa, 2021-07-24:
I could reproduce the problem with the last program using Thonny 3.3.11:

```
INFO    thonny: Thonny version: 3.3.11
INFO    thonny.plugins.micropython.bare_metal_backend: Choosing raw paste submit mode
INFO    thonny.plugins.micropython.backend: Prepared
INFO    thonny.plugins.rpi_pico.rpi_pico_backend: Syncing time in Pico
INFO    thonny.plugins.micropython.backend: Handling command 'get_dirs_children_info'
INFO    thonny.plugins.micropython.backend: Handling command 'get_globals'
INFO    thonny.plugins.micropython.backend: Handling command 'Run'
INFO    thonny.plugins.micropython.bare_metal_backend: WARNING: Could not use expected raw paste, falling back to raw
INFO    thonny.plugins.micropython.backend: Caught ManagementError
Traceback (most recent call last):
  File "/home/annamaa/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 343, in _handle_normal_command
    response = handler(cmd)
  File "/home/annamaa/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 977, in _cmd_Run
    self._prepare_after_soft_reboot(False)
  File "/home/annamaa/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 154, in _prepare_after_soft_reboot
    self._execute_without_output(script)
  File "/home/annamaa/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 554, in _execute_without_output
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:

class __thonny_helper:
    import builtins
    try:
        import uos as os
    except builtins.ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = builtins.dict()
    @builtins.classmethod
    def print_repl_value(cls, obj):
        global _
        if obj is not None:
            cls.builtins.print('[object_link_for_thonny=%d]' % cls.builtins.id(obj), cls.builtins.repr(obj), '[/object_link_for_thonny]', sep='')
            _ = obj

    @builtins.classmethod
    def print_mgmt_value(cls, obj):
        cls.builtins.print('<thonny>', cls.builtins.repr(obj), '</thonny>', sep='', end='')

    @builtins.classmethod
    def repr(cls, obj):
        try:
            s = cls.builtins.repr(obj)
            if cls.builtins.len(s) > 50:
                s = s[:50] + "..."
            return s
        except cls.builtins.Exception as e:
            return "<could not serialize: " + __thonny_helper.builtins.str(e) + ">"

    @builtins.classmethod
    def listdir(cls, x):
        if cls.builtins.hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @builtins.classmethod
    def getcwd(cls):
        if cls.builtins.hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @builtins.classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @builtins.classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:
Gluckwunsch asdf
Deine Reaktionszeit war 349 Millisekunden!

STDERR:

INFO    thonny.plugins.micropython.backend: Handling command 'get_globals'
```

aivarannamaa, 2021-07-24:
The output from the IRQ happened just when Thonny was preparing for the next management session and it thought that the setup command failed (because it is expected to run without output). I guess I need to make Thonny less strict about inspecting the results of a management command.

aivarannamaa, 2021-07-24:
Another possibility to fail -- the output may happen when Thonny is waiting for the command confirmation:

```
INFO    thonny.plugins.micropython.bare_metal_backend: WARNING: Could not use expected raw paste, falling back to raw
ERROR   thonny.plugins.micropython.backend: PROBLEM WITH THONNY'S BACK-END:
Traceback (most recent call last):
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/bare_metal_backend.py", line 557, in _submit_code
    self._submit_code_via_raw_paste_mode(to_be_sent)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/bare_metal_backend.py", line 643, in _submit_code_via_raw_paste_mode
    raise RawPasteNotSupportedError()
thonny.plugins.micropython.bare_metal_backend.RawPasteNotSupportedError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/backend.py", line 347, in _handle_normal_command
    response = handler(cmd)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/bare_metal_backend.py", line 958, in _cmd_Run
    self._prepare_after_soft_reboot(False)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/backend.py", line 155, in _prepare_after_soft_reboot
    self._execute_without_output(script)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/backend.py", line 591, in _execute_without_output
    out, err = self._execute(script, capture_output=False)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/backend.py", line 565, in _execute
    self._execute_with_consumer(script, self._send_output)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/bare_metal_backend.py", line 698, in _execute_with_consumer
    self._submit_code(script)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/bare_metal_backend.py", line 560, in _submit_code
    self._submit_code_via_raw_mode(to_be_sent)
  File "/home/annamaa/python_stuff/thonny/thonny/plugins/micropython/bare_metal_backend.py", line 624, in _submit_code_via_raw_mode
    raise AssertionError(
AssertionError: Could not read command confirmation. Got bytearray(b'Gluckwunsch Aivar\r\nDeine Reaktionszeit war 222 Millisekunden!\r\nOK')

SCRIPT:
b'\nclass __thonny_helper:\n    import builtins\n    try:\n        import uos as os\n    except builtins.ImportError:\n        import os\n    import sys\n\n    # for object inspector\n    inspector_values = builtins.dict()\n    @builtins.classmethod\n    def print_repl_value(cls, obj):\n        global _\n        if obj is not None:\n            cls.builtins.print(\'[object_link_for_thonny=%d]\' % cls.builtins.id(obj), cls.builtins.repr(obj), \'[/object_link_for_thonny]\', sep=\'\')\n            _ = obj\n\n    @builtins.classmethod\n    def print_mgmt_value(cls, obj):\n        cls.builtins.print(\'<thonny>\', cls.builtins.repr(obj), \'</thonny>\', sep=\'\', end=\'\')\n\n    @builtins.classmethod\n    def repr(cls, obj):\n        try:\n            s = cls.builtins.repr(obj)\n            if cls.builtins.len(s) > 50:\n                s = s[:50] + "..."\n            return s\n        except cls.builtins.Exception as e:\n            return "<could not serialize: " + __thonny_helper.builtins.str(e) + ">"\n\n    @builtins.classmethod\n    def listdir(cls, x):\n        if cls.builtins.hasattr(cls.os, "listdir"):\n            return cls.os.listdir(x)\n        else:\n            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in (\'.\', \'..\')]\n\n\n    @builtins.classmethod\n    def getcwd(cls):\n        if cls.builtins.hasattr(cls, "getcwd"):\n            return cls.os.getcwd()\n        else:\n            # micro:bit\n            return ""\n\n    @builtins.classmethod\n    def chdir(cls, x):\n        return cls.os.chdir(x)\n\n    @builtins.classmethod\n    def rmdir(cls, x):\n        return cls.os.rmdir(x)\n'
INFO    thonny.plugins.micropython.backend: Handling command 'get_globals'
DEBUG   thonny.plugins.micropython.bare_metal_backend: Submitting via raw_paste: b'__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value),'
DEBUG   thonny.plugins.micropython.backend: Returning 10 globals
```

aivarannamaa, 2021-07-25:
Non-ascii part of the problem got a remedy in #1911 the main part will be dealt with under #1913

closed 2021-07-25 by aivarannamaa