thonny/thonny #1778: Feature suggestion: command line argument to set the port

toybuilder, 2021-04-22:
My son started using Thonny to learn Python -- thanks for this wonderful program!

When he showed it to me, I decided to use it for my own projects, programming on Micropython running on multiple target boards at the same time.  I have set my Thonny configuration so that I can have a separate instance for each target board, which works fairly well, except that I have to manually set the interpreter's port connection.

I think it would be useful to be able to pass arguments to Thonny so that different instances can be pointed to their respective serial ports.  An additional nice-to-have would be to have the active port info displayed alongside the interpreter (target device) name.

aivarannamaa, 2021-04-25:
Thank you for the suggestion!

It looks quite similar to another suggestion: #1695 and I hope your case can be also solved by #1703.

toybuilder, 2021-04-25:
Oh, yes - it's definitely , I think #1703 sounds great,  

Just to provide additional feedback, I "decorate" the two different windows with contrasting color scheme so that, in addition to the window position, the color will quickly remind me of which unit I am working on. 

BTW, I've noticed that a Thonny window will resize/move to a different location sometimes (I think having to do exceptions that arise when the interpreter interface goes awry).  I don't know enough to know if the repositioning is deliberate or some unintended behavior, but if this is something that could easily be turned off with a command line flag, I think that would be nice.   

And, one more thought, if I may -- if it's possible to pass in a optional config file (i.e. "-c session.conf"), it would make it even easier to use by making changes in the .conf file instead of having to fix up scripts and launch icons.

Many thanks!

aivarannamaa, 2021-04-28:
> BTW, I've noticed that a Thonny window will resize/move to a different location sometimes (I think having to do exceptions that arise when the interpreter interface goes awry). I don't know enough [...]

I've had reports about this, but I haven't been able to reproduce it so far. If you find a reliable way to reproduce it then please let me know!

I'm considering the alternative configuration files suggestion.

toybuilder, 2021-04-29:
It happens somewhat regularly for me when I am jumping between multiple boards while developing on custom ESP32 targets, all running the stable Micropython 1.14 release.   Since that's the only setup I have used so far, it is not clear to me whether the issue exists with other versions or other platforms.

The problem seems to crop up the most if I am trying to interact with the target after a power cycles and if I issue back-to-back ctrl-C's to or use the start/restart while a program is running (versus being in the REPL).   I will try to catch a screenshot/videograb the next time to better document this.

toybuilder, 2021-05-07:
Ah, I've figured out the window sizing problem: #1811

closed 2021-06-14 by aivarannamaa